JuliaRobotics/RoME.jl #35: Getting RoME Running on ARM

Affie, 2017-11-20:
Trying RoME on raspberry pi 3 Rasbian Stretch with julia 0.61.
Pkg.build succeeds but,
Errors while trying to precompile:
Cairo.jl UndefVarError: _jl_libcairo not defined
Failed to precompile Compose, Gadfly, KernelDensityEstimate

Affie, 2017-11-22:
Thanks for separating plotting functions, so far its working for KernelDensityEstimates.jl

dehann, 2017-11-30:
Hi,

Thanks, will merge #37 soon, after registering [RoMEPlotting.jl](http://www.github.com/dehann/RoMEPlotting.jl) and [KernelDensityEstimatePlotting.jl](http://www.github.com/dehann/KernelDensityEstimatePlotting.jl).

Please also note PRs:
- [KernelDensityEstimatePlotting.jl#1](http://github.com/dehann/KernelDensityEstimatePlotting.jl/pull/1)
- [IncrementalInference.jl#73](http://github.com/dehann/IncrementalInference.jl/pull/73)

Affie, 2017-12-14:
Pkg.test("RoME") on pi. All test pass!

So separated packages works, plots can be fixed by fixing Cairo.jl for rome

closed 2017-12-14 by Affie