JuliaRobotics/RoME.jl #36: String issue

GearsAD, 2017-11-23:
Hi,

I'm having issues using the addOdoFG! function, mostly because of the AbstractString generic (think that's the term?). I've modified my version to this, which works:

RobotUtils.jl:162 - 
 
function addOdoFG!(
        fg::FactorGraph,
        n::Symbol,
        DX::Array{Float64,1},
        cov::Array{Float64,2};
        N::Int=0,
        ready::Int=1,
        labels::Vector{String}=String[]  )
    #
    prev, X, nextn = getLastPose2D(fg)
    r,c = size(X)
    if N==0
      N = c
    end
    sig = diag(cov)
    RES = zeros(r,c)
    # increases the number of particles based on the number of modes in the measurement Z
    for i in 1:c
        ent = [randn()*sig[1]; randn()*sig[2]; randn()*sig[3]]
        RES[:,i] = addPose2Pose2(X[:,i], DX + ent)
    end

    v = addNode!(fg, n, RES, cov, N=N, ready=ready, labels=labels)
    pp = Pose2Pose2(vectoarr2(DX), cov, [1.0]) #[prev;v],
    f = addFactor!(fg, [prev;v], pp, ready=ready)
    infor = inv(cov^2)
    # addOdoRemote(prev.index,v.index,DX,infor) # this is for remote factor graph ref parametric solution -- skipped internally by global flag variable
    return v, f
end

dehann, 2017-11-30:
HI @GearsAD ,

Sure, which kind of issues are you seeing? Would you mind trying this instead please, and let us know if that works:
```julia
function addOdoFG!(
        fg::FactorGraph,
        n::Symbol,
        DX::Array{Float64,1},
        cov::Array{Float64,2};
        N::Int=0,
        ready::Int=1,
        labels::Vector{<:AbstractString}=String[]  )
    #
    prev, X, nextn = getLastPose2D(fg)
    r,c = size(X)
    if N==0
      N = c
    end
    sig = diag(cov)
    RES = zeros(r,c)
    # increases the number of particles based on the number of modes in the measurement Z
    for i in 1:c
        ent = [randn()*sig[1]; randn()*sig[2]; randn()*sig[3]]
        RES[:,i] = addPose2Pose2(X[:,i], DX + ent)
    end

    v = addNode!(fg, n, RES, cov, N=N, ready=ready, labels=labels)
    pp = Pose2Pose2(vectoarr2(DX), cov, [1.0]) #[prev;v],
    f = addFactor!(fg, [prev;v], pp, ready=ready)
    infor = inv(cov^2)
    # addOdoRemote(prev.index,v.index,DX,infor) # this is for remote factor graph ref parametric solution -- skipped internally by global flag variable
    return v, f
end
```

dehann, 2017-11-30:
FYI:
```julia
julia> function ff(st::Vector{<:AbstractString})
         @show st
       end
ff (generic function with 1 method)

julia> s = String["hello"; "world"]
2-element Array{String,1}:
 "hello"
 "world"

julia> ff(s)
st = String["hello", "world"]
2-element Array{String,1}:
 "hello"
 "world"
```

GearsAD, 2018-04-25:
Yep, that works! https://github.com/dehann/RoME.jl/blob/master/src/RobotUtils.jl#L165-L193

closed 2018-04-25 by GearsAD