JuliaRobotics/RoME.jl #113: Pose2Point2Range throws error when adding it to a graph

GearsAD, 2018-09-16:
This is at BearingRange2D, here:
  z = pp2r.Zij[1]+meas[1][1,idx]

Should it be   z = pp2r.Z[1]+meas[1][1,idx] ?

Stacktrace:
type Pose2Point2Range has no field Zij
in anonymous at base/<missing>
in macro expansion at Sandshark_2bcontinued.jl:163 
in addFactor! at IncrementalInference/src/FactorGraph01.jl:634
in #addFactor!#53 at IncrementalInference/src/FactorGraph01.jl:638
in  at base/<missing>
in #addFactor!#52 at IncrementalInference/src/FactorGraph01.jl:609
in  at base/<missing>
in #doautoinit!#45 at IncrementalInference/src/FactorGraph01.jl:497
in  at base/<missing>
in #predictbelief#136 at IncrementalInference/src/SolveTree01.jl:338
in  at base/<missing>
in #predictbelief#135 at IncrementalInference/src/SolveTree01.jl:313
in  at base/<missing>
in #proposalbeliefs!#134 at IncrementalInference/src/SolveTree01.jl:282
in findRelatedFromPotential at IncrementalInference/src/ApproxConv.jl:381
in  at base/<missing>
in #evalFactor2#130 at IncrementalInference/src/ApproxConv.jl:333
in  at base/<missing>
in #evalPotentialSpecific#127 at IncrementalInference/src/ApproxConv.jl:225
in computeAcrossHypothesis! at IncrementalInference/src/ApproxConv.jl:125
in approxConvOnElements! at IncrementalInference/src/ApproxConv.jl:68
in approxConvOnElements! at IncrementalInference/src/ApproxConv.jl:46
in #numericRootGenericRandomizedFnc!#114 at IncrementalInference/src/SolverUtilities.jl:94
in #nlsolve#39 at NLsolve/src/nlsolve/nlsolve.jl:59
in  at base/<missing>
in #nlsolve#38 at NLsolve/src/nlsolve/nlsolve.jl:26
in trust_region_ at NLsolve/src/solvers/trust_region.jl:118
in value_jacobian!! at NLSolversBase/src/interface.jl:124

dehann, 2018-09-16:
`pp2r.Zij[1]+` does not seem right.  Try without since it should not be there.

GearsAD, 2018-09-28:
This is fixed in https://github.com/JuliaRobotics/RoME.jl/tree/hotfix/fixpose2point2brsample

closed 2018-09-30 by dehann