JuliaRobotics/RoME.jl #173: tag v0.3.4

dehann, 2019-05-02:
@JuliaRegistrator register()

JuliaRegistrator, 2019-05-02:
Registration pull request created: [JuliaRegistries/General/489](https://github.com/JuliaRegistries/General/pull/489)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.
    
This will be done automatically if [Julia TagBot](https://github.com/apps/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.3.4 -m "<description of version>" 5ce9fb8b244645fc2e0e153284c7fd7f40dec0f2
git push origin v0.3.4
```

dehann, 2019-05-02:
@JuliaRegistrator register()

JuliaRegistrator, 2019-05-02:
Registration pull request updated: [JuliaRegistries/General/489](https://github.com/JuliaRegistries/General/pull/489)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.
    
This will be done automatically if [Julia TagBot](https://github.com/apps/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.3.4 -m "<description of version>" 34733207a01dc07654e941efa919d2db569022a0
git push origin v0.3.4
```

dehann, 2019-05-02:
https://github.com/JuliaRobotics/RoME.jl/releases/tag/v0.3.4

closed 2019-05-02 by dehann