JuliaRobotics/RoME.jl #184: preallocate rotation matrix for Pose2 operations

dehann, 2019-07-01:
A test suggests there are 18 allocations within each `AMP.difftheta` operation.

mc2922, 2019-07-01:
This should be 0 allocations for difftheta, although requires preallocating the transform matrix:

```
function difftheta2(wth1::Float64, wth2::Float64, Rtemp::Matrix{Float64})
    Rtemp[1,1] = cos(wth1)*cos(wth2)+sin(wth1)*sin(wth2)
    Rtemp[1,2] = cos(wth1)*-sin(wth2)+sin(wth1)*cos(wth2)
    Rtemp[2,1] = -sin(wth1)*cos(wth2)+cos(wth1)*sin(wth2)
    Rtemp[2,2] = -sin(wth1)*-sin(wth2)+cos(wth1)*cos(wth2)
    logmap_SO2(Rtemp)
end
```

dehann, 2019-07-01:
Awesome thanks, I will add this soon!

Affie, 2022-08-10:
I think this is no longer relevant with Manifolds.jl upgrade.  The current plan is to work towards static arrays for zero allocations.

closed 2022-08-10 by Affie