JuliaRobotics/RoME.jl #189: Bounds wrong in julia registry

Affie, 2019-08-08:
The registry seems to have messed up with the bounds for IIF in [compat](https://github.com/JuliaRegistries/General/blob/ccfe8562495b5f3cd6ce7ebaaadbf0249a4c18a4/R/RoME/Compat.toml#L167) 
Not sure how we want to fix it?
- new tag or
- pr against registry to make it IncrementalInference = "0.7.2-0.8"

There seems to be a lot of confusion/arguing about the bounds, see
https://github.com/JuliaLang/Pkg.jl/issues/1177#event-2304129269
https://discourse.julialang.org/t/dependency-problem-with-fillarrays/

dehann, 2019-08-08:
Maybe we should change that to
compat = “^0.7.2”

Affie, 2019-08-08:
As I understand documentation it `"0.7.2"` should default to `^`. I looked at other packages that had similar requirements and they looked fine

dehann, 2019-08-08:
I wonder where those compat requirements changes, because they were different a while back...

dehann, 2019-08-08:
It’s unclear how much IIF v0.8.x is going to change from v0.7.x at this point so really want to box RoME dependency on IIF to only v0.7.x

Affie, 2019-08-08:
Something between documentation and implementation is mismatched.

dehann, 2019-08-08:
What did you change on your registry to fix it.  Did you just change the Project.toml with 0.7.2 or something else in the actual registry files too?

dehann, 2019-08-08:
Hold on I see now that RoME.jl/Project.toml has compat with IIF = "^0.7.2", which is right according to docs.  Agree, this really is a Pkg issue then?

Affie, 2019-08-08:
Look at the dependency failures on https://travis-ci.org/JuliaRobotics/DistributedFactorGraphs.jl/builds/568981621
It passes on 1.0 and fails on 1.1.
I think its either Pkg or Registry, I thought the second number was excluded eg. [v0.1.3,v0.2) Thats why I thought it may be registry, but don't know

dehann, 2019-08-08:
Okay, i missed that second number is excluded.  I thought the second number is precisely included, or is that only when doing IIF = "0.72, 0.8".  So we want to do IIF = "0.7.2-0.8" but I dont think we can write that in RoME/Project.toml i think?

dehann, 2019-08-08:
So there is an issue with both DFG and RoME then?

Affie, 2019-08-08:
It might have changed with toml, was like that with requires

dehann, 2019-08-08:
Leave it open till figure it out

dehann, 2019-08-08:
I thought the tomls were right and it used to work fine... I'm a little confused :-/

dehann, 2019-08-08:
Hi @GearsAD , any chance that new DFG 0.3 broke the registry?  We having failures from DFG thorugh RoME.  I'm going to try fix IIF quick, think the manifest is behind

Might be the culprit:
https://github.com/JuliaRegistries/General/pull/2557

dehann, 2019-08-08:
see here: https://travis-ci.org/JuliaRobotics/DistributedFactorGraphs.jl

dehann, 2019-08-08:
Checking new IIF tests for the same dependency bug here:
https://travis-ci.org/JuliaRobotics/IncrementalInference.jl

Affie, 2019-08-08:
I think something strange is going on. if you look in the lists they overlap. but say no versions left. Not a RoME thing

dehann, 2019-08-08:
IIF is passing but IIF is also restricted to only use DFG v0.2.x

dehann, 2019-08-08:
Think the issue is RoME and IIF are mutually exclusive in their dependency on DFG:

From DFG travis with unintended IIF and RoME dependency:
```
restricted by compatibility requirements with DistributedFactorGraphs [b5cc3c7e] to versions: [0.0.1-0.0.2, 0.1.0, 0.2.0-0.2.11, 0.3.0-0.3.9, 0.4.0-0.4.7, 0.5.0-0.5.7, 0.6.0, 0.7.0] or uninstalled — no versions left
```

dehann, 2019-08-22:
I just removed the manifest toml here in RoME, and IIF v0.7.5 has been tagged so that should work...

dehann, 2019-09-28:
closing for now

closed 2019-09-28 by dehann