JuliaRobotics/RoME.jl #202: tag v0.4.3

dehann, 2019-10-17:
@JuliaRegistrator register

JuliaRegistrator, 2019-10-17:
Registration pull request created: [JuliaRegistries/General/4449](https://github.com/JuliaRegistries/General/pull/4449)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if [Julia TagBot](https://github.com/apps/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.4.3 -m "<description of version>" 9b32e22b7f12c039d758cdb67a24556fa6b1ff93
git push origin v0.4.3
```

dehann, 2019-10-17:
@JuliaRegistrator register

JuliaRegistrator, 2019-10-17:
Registration pull request updated: [JuliaRegistries/General/4449](https://github.com/JuliaRegistries/General/pull/4449)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if [Julia TagBot](https://github.com/apps/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.4.3 -m "<description of version>" 0980897620421d3b28fd20d6c016e4c091cf126d
git push origin v0.4.3
```