JuliaRobotics/RoME.jl #244: Integrate with Manifolds.jl

dehann, 2020-02-20:
This also affects ApproxManifoldProducts.jl directly.

- E(n), https://julianlsolvers.github.io/Manifolds.jl/latest/manifolds/euclidean.html
- SO(n), https://julianlsolvers.github.io/Manifolds.jl/latest/manifolds/rotations.html
  - Quaternions, https://julianlsolvers.github.io/Manifolds.jl/latest/lib/numbers.html#Manifolds.QuaternionNumbers
- SE(n), https://julianlsolvers.github.io/Manifolds.jl/latest/manifolds/group.html#Special-Euclidean-group-1
-

dehann, 2020-02-20:
Retroactively influences JuliaRobotics/DistributedFactorGraphs.jl#147

dehann, 2020-02-20:
@Affie, if I understand your earlier comment right -- Optim.jl currently has support for Sphere or Euclidean and other Manifolds.jl but not SE(n) (Special Euclidean) yet?

Affie, 2020-02-20:
> Optim.jl currently has support for Sphere or Euclidean and other Manifolds.jl but not SE(n) (Special Euclidean) yet?

Optim.jl does not support Manifolds.jl yet as far as I can tell.

> SE(n)

Thanks, I missed SE in Manifolds.jl, I didn't think to look in decorators.

Affie, 2020-02-20:
https://github.com/JuliaNLSolvers/Manifolds.jl/issues/35

kellertuer, 2020-02-20:
Cool, that you already have this in your plans!

Some simple integration to `Optim.jl` will happen at some time, for example a gradient descent is quite straight forward.
(Biased side note): For more advanced algorithms doing Optimization on Manifold, see [Manopt.jl](https://manoptjl.org), which will also switch to be based on `Manifolds.jl` as soon as there is a public release.

dehann, 2020-05-13:
JuliaManifolds/Manopt.jl#24

dehann, 2020-06-19:
https://discourse.julialang.org/t/understanding-macros-a-macro-to-decorate-a-type/12204

dehann, 2020-10-19:
xref #341

dehann, 2020-11-21:
xref, need to rebuild densities starting from JuliaRobotics/ApproxManifoldProducts.jl#41

dehann, 2021-03-29:
v0.15.0 closes this issue (from RoME.jl perspective).  Major future work remains deeper inside and will be tracked at JuliaRobotics/ApproxManifoldProducts.jl#41

dehann, 2021-03-29:
PS, the local manifolds, `SE2E2_Manifold` and `BearingRange_Manifold` are not yet consolidated with the standard Manifolds.jl interface format.  That work still remains to be done.  The main delivery here is that all variables and surrounding code from RoME IIF AMP not expect and use `<:Manifold`.

closed 2021-03-29 by dehann