JuliaRobotics/RoME.jl #334: Looking for a simple, pure inertial motion estimation

pauljurczak, 2020-09-17:
Is there a way to get a simple, pure inertial motion estimation with RoME.jl o rrelated packages? Ultimate accuracy is not required as I want to use it for motion estimation between camera frames capture events, every 50ms with IMU data rate of 250Hz.

dehann, 2020-09-18:
Hi @pauljurczak ,  yes and no answer -- yes those factors exist but we are not able to share them publicly just yet owing to license restrictions.  We are working on getting those available but will still take a bit of time to get that legally sound.  Wish we could make them immediately available.  In the mean time we are working on integrating `IncrementalInference.jl` with `Manifolds.jl` so that 3D / IMU-derived (non-Gaussian/multimodal) beliefs, and their underlying mechanics, are well organized and performant -- the combination should be pretty good.

xref: #244

pauljurczak, 2020-09-19:
Thank you for responding. In the meantime, do you know another Julia package or C++ library, which does pure inertial motion estimation? Everything I found so far fuses inertial with visual or other sensing modes.

dehann, 2020-10-19:
Apologies for the slow reply.  Perhaps the fastest thing to do is just embed an ODE of the classic inertial navigation solution (aking to *preintegrating*).  This will provide similar or better numerical answers with a little computation slowdown -- it will, however, avoid licensing delays.

dehann, 2020-11-09:
Just an update on this, I'm busy adding `ODERelative` as a new factor that can stand up generic ODE solutions (discrete and consuming sensor data) as a stand in solution for this and other similar requirements.  See JuliaRobotics/IncrementalInference.jl#1020 for ongoing progress.  I want to write a regular and classic INS solution in as well, which will be able to solve the problem described above.  The native inertial pre-integral (ie more computationally efficient) version will be made public as soon as licensing is sorted out.

dehann, 2023-11-14:
see new on-manifold code for factors `InertialDynamic` and `IMUDeltaFactor`.  These functions work in part either with parametric or nonparametric solving.

This new functionality supercedes previous questions such as:
- #641  

Note that InertialPose3 may well be used in the future, but currently a new variable `RotVelPos` is being used as a stand-in solution until all features are consolidated in common.

Next major milestone towards fully consolidated operations  will be to complete
- JuliaRobotics/ApproxManifoldProducts.jl#41

closed 2023-11-14 by dehann