JuliaRobotics/RoME.jl #342: Example about semantic factors

Sanghyun-Cha, 2020-10-26:
I'm studying on semantic SLAM, and recently, I read following papers.

- Kevin Doherty, Dehann Fourie, John Leonard, “Multimodal Semantic SLAM with Probabilistic Data Association”, 2019 International Conference on Robotics and Automation (ICRA), 2019.
- Dehann Fourie, John Leonard, Michael Kaess. "A nonparametric belief solution to the Bayes tree." 2016 IEEE/RSJ International Conference on Intelligent Robots and Systems (IROS). IEEE, 2016.

Is there some example code about using semantic factors?

dehann, 2020-10-27:
Hi @Sanghyun-Cha , apologies for the slow reply -- will add docs regarding these examples soon and notify you here again.  In the meantime the main feature you will be interested in is e.g. 100% certain about `:x1` side of regular binary factor, but then fracctional certainty about what the association is on "`:l1`", mocked as `a,b,c` for illustration
```julia
addFactor!(fg, [:x1; :l1a; :l1b; :l1c], fct, multihypo=[1; 0.4; 0.5; 0.1])
```

There are a few things to tweak if interested but will explain once, clearly in docs.

There is a basic example here:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/master/test/testMultiHypo2Door.jl

There is some really old code here, but perhaps useful for reference (note looking at this again though, its really old so don't go too far down that path).  Better to wait for the documentation to come.  Here is the really old code:
https://github.com/JuliaRobotics/Caesar.jl/blob/master/examples/wheeled/vicprk/VicPrkEstimator.jl

Will be best to use the latest IncrementalInference.jl v0.17.0 and RoME.jl v0.10.0 which we will be tagging in the next couple of days so just hang tight on that too or use `] add RoME#master` and `] add IncrementalInference#master`.  You can free back to stable package tags with `] add RoME` later.

Hi @keevindoherty,

Not sure if you are perhaps able to share code relating to the semantic slam paper 2019 not yet available.  I recall some simulation code here:
https://github.com/JuliaRobotics/Caesar.jl/tree/master/examples/simulation

Thanks either way!
Dehann

Sanghyun-Cha, 2020-10-29:
Thank you for your reply! It helped me a lot.

closed 2020-10-29 by Sanghyun-Cha

dehann, 2020-11-04:
Hi @Sanghyun-Cha , here is some more documentation following this issue:
https://juliarobotics.org/Caesar.jl/latest/concepts/dataassociation/

Sanghyun-Cha, 2020-11-06:
Thanks for the information!