JuliaRobotics/RoME.jl #364: tag v0.11.1

dehann, 2020-12-10:
@JuliaRegistrator register branch=releasev0.11

JuliaRegistrator, 2020-12-10:
Error while trying to register: "Branch `releasev0.11` not found"

dehann, 2020-12-10:
@JuliaRegistrator register branch=release/v0.11

JuliaRegistrator, 2020-12-10:
Registration pull request created: [JuliaRegistries/General/26157](https://github.com/JuliaRegistries/General/pull/26157)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.11.1 -m "<description of version>" fedfff5574afcaba2b475e618a5d9de4851e0521
git push origin v0.11.1
```

closed 2020-12-10 by dehann