JuliaRobotics/RoME.jl #382: Implement getParametricField where needed

Affie, 2021-01-21:
ERROR: LoadError: type Pose2Point2BearingRange has no field Z

I would guess that we need the new `getParametricField` for all non `.Z` factors 

- [ ] Pose2Point2BearingRange
https://github.com/JuliaRobotics/RoME.jl/blob/83f9b2f8b3cd60219c00370439024bc04166e40c/src/factors/BearingRange2D.jl#L10-L13

dehann, 2021-01-22:
Check what i did in IIF, assuming you trying to get mean and cov?

dehann, 2021-01-22:
either way, i think best is to get parametric to use the same factor (even if helpers like getParametricField are needed) just to get consolidation doen and close IIF 467.  Likely multiple rounds of improvements to do here, so let's do baby steps by consolidate one each?

Affie, 2021-01-28:
We are going with extending `getParametricMeasurement` instead, see #384

Affie, 2021-02-01:
#393 implements and tests some factors, rest can be done as needed.

closed 2021-02-01 by Affie