JuliaRobotics/RoME.jl #388: Error using Pose2Point2 Factor

lemauee, 2021-01-28:
Hi,

 For comparison to the Pose2Point2BearingRange factor, I wanted to try the Pose2Point2 factor on my well-known example:
 
```julia
## Read Gt Input from files

using CSV

associationLm = CSV.File("association_lm.txt", header=["iPose", "jLm"])
gtLm = CSV.File("gt_lm.txt", header=["x", "y", "θ"])
gtTraj = CSV.File("gt_traj.txt", header=["x", "y", "θ"])

## Generate "xy" measurements from groundtruth

using CoordinateTransformations, TransformUtils, LinearAlgebra

Tinv(x,y,θ) = inv(Translation(x,y) ∘ LinearMap(R(θ)))

mTrajGt = [Vector{Float64}(undef,3) for _ = 1:length(gtTraj)]
mLmGt = [Dict{Int,Vector{Float64}}() for _ = 1:length(gtTraj)]

for kPose = 1:length(gtTraj)

    # odometry
    if kPose > 1 # no odemetry for first pose, prior instead
        trans = Tinv(gtTraj[kPose-1].x,gtTraj[kPose-1].y,gtTraj[kPose-1].θ)([gtTraj[kPose].x, gtTraj[kPose].y])
        rot = wrapRad(gtTraj[kPose].θ - gtTraj[kPose-1].θ)
        mTrajGt[kPose] = vcat(trans,rot)
    end

    # landmark measurements
    lmIdx = associationLm.jLm[findall(associationLm.iPose .== kPose)]
    for kLm = lmIdx
        mLmGt[kPose][kLm] = Tinv(gtTraj[kPose].x,gtTraj[kPose].y,gtTraj[kPose].θ)([gtLm[kLm].x, gtLm[kLm].y])
    end
end

## Add noise to ground truth measurements

using Distributions, Random

Random.seed!(42); # reproducibility

ΣOdo = Diagonal([0.01,0.01,0.5*pi/180].^2)
ΣLm = Diagonal([0.01,0.01].^2)
# ΣOdo = Diagonal([0.1,0.1,5*pi/180].^2)
# ΣLm = Diagonal([0.1,0.1].^2)

noiseOdo = MvNormal([0,0,0],ΣOdo)
noiseLm = MvNormal([0,0],ΣLm)

mTrajNoisy = [Vector{Float64}(undef,3) for _ = 1:length(gtTraj)]
mLmNoisy = [Dict{Int,Vector{Float64}}() for _ = 1:length(gtTraj)]

for kPose = 1:length(gtTraj)

    # odometry
    if kPose > 1 # no odometry for first pose, prior instead
        mTrajNoisy[kPose] = mTrajGt[kPose] + vec(rand(noiseOdo,1))
    end

    # landmark measurements
    for kLm = keys(mLmGt[kPose])
        mLmNoisy[kPose][kLm] = mLmGt[kPose][kLm] + vec(rand(noiseLm,1))
    end
end

## Build fg from measurements and solve incrementally

[... 100 lines cut ...]
 [27] ensureAllInitialized!(::LightDFG{SolverParams,DFGVariable,DFGFactor}; solvable::Int64) at /home/leopold/.julia/packages/IncrementalInference/ShzPg/src/FactorGraph.jl:1054
 [28] ensureAllInitialized! at /home/leopold/.julia/packages/IncrementalInference/ShzPg/src/FactorGraph.jl:1039 [inlined]
 [29] solveTree!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree; timeout::Nothing, storeOld::Bool, verbose::Bool, verbosefid::Base.TTY, delaycliqs::Array{Symbol,1}, recordcliqs::Array{Symbol,1}, limititercliqs::Array{Pair{Symbol,Int64},1}, injectDelayBefore::Nothing, skipcliqids::Array{Symbol,1}, eliminationOrder::Nothing, variableOrder::Nothing, eliminationConstraints::Array{Symbol,1}, variableConstraints::Nothing, smtasks::Array{Task,1}, dotreedraw::Array{Int64,1}, runtaskmonitor::Bool, algorithm::Symbol, multithread::Bool) at /home/leopold/.julia/packages/IncrementalInference/ShzPg/src/SolverAPI.jl:307
 [30] solveTree!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree) at /home/leopold/.julia/packages/IncrementalInference/ShzPg/src/SolverAPI.jl:280
 [31] top-level scope at /home/leopold/Nextcloud/Studium/Masterarbeit/svn/julia/mmiSAM/examples/br_issue/br_issue_pose2point2.jl:108
 [32] include_string(::Function, ::Module, ::String, ::String) at ./loading.jl:1088
 [33] include_string(::Module, ::String, ::String) at ./loading.jl:1096
 [34] invokelatest(::Any, ::Any, ::Vararg{Any,N} where N; kwargs::Base.Iterators.Pairs{Union{},Union{},Tuple{},NamedTuple{(),Tuple{}}}) at ./essentials.jl:710
 [35] invokelatest(::Any, ::Any, ::Vararg{Any,N} where N) at ./essentials.jl:709
 [36] inlineeval(::Module, ::String, ::Int64, ::Int64, ::String; softscope::Bool) at /home/leopold/.vscode-oss/extensions/julialang.language-julia-1.0.10/scripts/packages/VSCodeServer/src/eval.jl:185
 [37] (::VSCodeServer.var"#61#65"{String,Int64,Int64,String,Module,Bool,VSCodeServer.ReplRunCodeRequestParams})() at /home/leopold/.vscode-oss/extensions/julialang.language-julia-1.0.10/scripts/packages/VSCodeServer/src/eval.jl:144
 [38] withpath(::VSCodeServer.var"#61#65"{String,Int64,Int64,String,Module,Bool,VSCodeServer.ReplRunCodeRequestParams}, ::String) at /home/leopold/.vscode-oss/extensions/julialang.language-julia-1.0.10/scripts/packages/VSCodeServer/src/repl.jl:124
 [39] (::VSCodeServer.var"#60#64"{String,Int64,Int64,String,Module,Bool,Bool,VSCodeServer.ReplRunCodeRequestParams})() at /home/leopold/.vscode-oss/extensions/julialang.language-julia-1.0.10/scripts/packages/VSCodeServer/src/eval.jl:142
 [40] hideprompt(::VSCodeServer.var"#60#64"{String,Int64,Int64,String,Module,Bool,Bool,VSCodeServer.ReplRunCodeRequestParams}) at /home/leopold/.vscode-oss/extensions/julialang.language-julia-1.0.10/scripts/packages/VSCodeServer/src/repl.jl:36
 [41] (::VSCodeServer.var"#59#63"{String,Int64,Int64,String,Module,Bool,Bool,VSCodeServer.ReplRunCodeRequestParams})() at /home/leopold/.vscode-oss/extensions/julialang.language-julia-1.0.10/scripts/packages/VSCodeServer/src/eval.jl:110
 [42] with_logstate(::Function, ::Any) at ./logging.jl:408
 [43] with_logger at ./logging.jl:514 [inlined]
 [44] (::VSCodeServer.var"#58#62"{VSCodeServer.ReplRunCodeRequestParams})() at /home/leopold/.vscode-oss/extensions/julialang.language-julia-1.0.10/scripts/packages/VSCodeServer/src/eval.jl:109
 [45] #invokelatest#1 at ./essentials.jl:710 [inlined]
 [46] invokelatest(::Any) at ./essentials.jl:709
 [47] macro expansion at /home/leopold/.vscode-oss/extensions/julialang.language-julia-1.0.10/scripts/packages/VSCodeServer/src/eval.jl:27 [inlined]
 [48] (::VSCodeServer.var"#56#57")() at ./task.jl:356
in expression starting at /home/leopold/Nextcloud/Studium/Masterarbeit/svn/julia/mmiSAM/examples/br_issue/br_issue_pose2point2.jl:80
```
For the additional files, see #380

Affie, 2021-01-29:
see backport v0.12.1 for fix

lemauee, 2021-02-03:
Sorry for the late update,

Its running now, but the results are pretty similar to whats going on with the Pose2Point2BearingRange Factor (#380):

Fine up to Pose ~60:
[image]

Messed up at Pose 150:
[image]

So this now also should be related to JuliaRobotics/IncrementalInference.jl#1010 and JuliaRobotics/IncrementalInference.jl#1051. Maybe opening a new issue for this, renaming and reopening this one or adding this example somwhere else is a good thing to keep track of it.

dehann, 2021-02-07:
Will investigate thanks, sorry we may have broken something here (it should not be this sensitive).  We just completed major internal refactoring of the code in IIF and RoME.  I think a common fix option would be to troubleshoot and then resolve JuliaRobotics/IncrementalInference.jl#1051 sooner rather than later.

Whats supposed to happen (as individual steps) is when solving a Point2 from a Pose2 through the factor should give a definitive result:
```julia
pp = Pose2Point2(..) 
f1 = addFactor!(fg, [:x1; :l1], pp)

# high to low dim
pts = approxConv(fg, f1.label, :l1)
```

The reverse should result in approx uniform spread of points in the dimensions that are not constrained by the factor:
```julia
pts_ = approxConv(fg, f1.label, :x1)
```

In this case the theta variable on `:x1` should be near uniform.

```julia
using RoMEPlotting

# might not be the precise API, im trying to recall without testing
plotPose(manikde!(pts_, Pose))
```

Should see even spread on theta, but x and y should be in line with Point2 x and y according to `pp`.

PR #389 (2021-01-31): Fix and test Pose2Point2
close #388 

we need to backport this also, see #390