JuliaRobotics/RoME.jl #396: Advice on using Dead Reckon Tether

lemauee, 2021-02-04:
As suggested, to speed up the computation and come closer to real time usage, I want to use the Dead Reckon Tether Approach presented in https://juliarobotics.org/Caesar.jl/latest/examples/deadreckontether/#Dead-Reckon-Tether and only solve the fg every n poses while still getting a good estimate for the current pose using DRT.

Are there any (simple) examples or more documentation on how the MutablePose2Pose2Gaussian factor and corresponding functions are used in such a setup? The best thing I could come up up with is this modification of my well-known example:

```julia
## Read Gt Input from files

using CSV

associationLm = CSV.File("association_lm.txt", header=["iPose", "jLm"])
gtLm = CSV.File("gt_lm.txt", header=["x", "y", "θ"])
gtTraj = CSV.File("gt_traj.txt", header=["x", "y", "θ"])

## Generate bearing range measurements from groundtruth

using CoordinateTransformations, TransformUtils, LinearAlgebra

Tinv(x,y,θ) = inv(Translation(x,y) ∘ LinearMap(R(θ)))

mTrajGt = [Vector{Float64}(undef,3) for _ = 1:length(gtTraj)]
mLmGt = [Dict{Int,Vector{Float64}}() for _ = 1:length(gtTraj)]

for kPose = 1:length(gtTraj)

    # odometry
    if kPose > 1 # no odemetry for first pose, prior instead
        trans = Tinv(gtTraj[kPose-1].x,gtTraj[kPose-1].y,gtTraj[kPose-1].θ)([gtTraj[kPose].x, gtTraj[kPose].y])
        rot = wrapRad(gtTraj[kPose].θ - gtTraj[kPose-1].θ)
        mTrajGt[kPose] = vcat(trans,rot)
    end

    # landmark measurements
    lmIdx = associationLm.jLm[findall(associationLm.iPose .== kPose)]
    for kLm = lmIdx
        trans = Tinv(gtTraj[kPose].x,gtTraj[kPose].y,gtTraj[kPose].θ)([gtLm[kLm].x, gtLm[kLm].y])
        rot = wrapRad(gtLm[kLm].θ - gtTraj[kPose].θ)
        mLmGt[kPose][kLm] = vcat(trans,rot)
    end
end

## Add noise to ground truth measurements

using Distributions, Random

Random.seed!(42); # reproducibility

ΣOdo = Diagonal([0.01,0.01,0.5*pi/180].^2)
ΣLm = Diagonal([0.01,0.01,0.5*pi/180].^2)
# ΣOdo = Diagonal([0.1,0.1,5*pi/180].^2)
# ΣLm = Diagonal([0.1,0.1,5*pi/180].^2)

noiseOdo = MvNormal([0,0,0],ΣOdo)
noiseLm = MvNormal([0,0,0],ΣLm)

mTrajNoisy = [Vector{Float64}(undef,3) for _ = 1:length(gtTraj)]
mLmNoisy = [Dict{Int,Vector{Float64}}() for _ = 1:length(gtTraj)]

for kPose = 1:length(gtTraj)

    # odometry
    if kPose > 1 # no odemetry for first pose, prior instead
        mTrajNoisy[kPose] = mTrajGt[kPose] + vec(rand(noiseOdo,1))
    end

    # landmark measurements
    for kLm = keys(mLmGt[kPose])
        mLmNoisy[kPose][kLm] = mLmGt[kPose][kLm] + vec(rand(noiseLm,1))
    end
end

## Build fg from measurements and solve incrementally

using Distributed, RoME
@everywhere using RoME

tRun = @elapsed begin
    global endPose = 10;
    global solveStep = 5;
    global fg = LightDFG{SolverParams}(solverParams=SolverParams(logpath="speed"))
    getSolverParams(fg).useMsgLikelihoods = false
    # getSolverParams(fg).isfixedlag = true
    # getSolverParams(fg).qfl = 50
    # getSolverParams(fg).limitfixeddown = true
    firstsolve = true
    drecPoses = Array{Float64,2}(undef,endPose,3)

    function addVarIfMissing!(fg,symbol,type,tags)
        if (symbol in ls(fg)) == false
            @info "adding missing variable $(symbol)"
            addVariable!(fg, symbol, type, tags = tags)
        end
    end

    for kPose = 1:endPose
        
        global fg,tree,smt,hist,drec,lastSolveIdx,firstsolve

        # odometry
        if kPose > 1 # no odometry for first pose
            fromPose = Symbol("x", kPose-1)
            toPose = Symbol("x", kPose)
            addVarIfMissing!(fg, fromPose, Pose2, [:POSE])
            addVarIfMissing!(fg, toPose, Pose2, [:POSE])   
            addFactor!(fg, [fromPose, toPose], Pose2Pose2(MvNormal(mTrajNoisy[kPose], ΣOdo)), graphinit = false)
            initManual!(fg, toPose, [Symbol("x",kPose-1,"x",kPose,"f",1)]) # doing this because I initlialize the same way in GTSAM
        else # prior instead
            addVariable!(fg, :x1, Pose2)
            addFactor!(fg, [:x1], PriorPose2(MvNormal([gtTraj[1].x, gtTraj[1].y, gtTraj[1].θ], Diagonal([0.01,0.01,0.5*pi/180].^2))))
        end

        # landmark measurements
        fromPose = Symbol("x", kPose)
        for kLm = collect(keys(mLmGt[kPose]))
            toLm = Symbol("l", kLm)
            addVarIfMissing!(fg, fromPose, Pose2, [:POSE])
            addVarIfMissing!(fg, toLm, Pose2, [:LANDMARK])
            addFactor!(fg, [fromPose, toLm], Pose2Pose2(MvNormal(mLmNoisy[kPose][kLm], ΣLm)))
        end

        # solve
        if mod(kPose,solveStep) == 1 # full solve
            if firstsolve
                tree, smt, hist = solveTree!(fg; multithread=true)
                firstsolve = false
            else
                tree, smt, hist = solveTree!(fg, tree; multithread=true)
            end
            drecSym = Symbol("deadreckon_x",kPose)
            addVariable!(fg, drecSym, Pose2, solvable=0)
            global drec = MutablePose2Pose2Gaussian(MvNormal(zeros(3), Matrix{Float64}(LinearAlgebra.I, 3,3)))               
            addFactor!(fg, [Symbol("x",kPose); drecSym], drec, solvable=0)
            global drecPoses[kPose,:] = accumulateFactorMeans(fg, [Symbol("x",drecIdx,"deadreckon_x",drecIdx,"f1")])
            lastSolveIdx = kPose
        else # DRT
            accumulateDiscreteLocalFrame!(drec, mTrajNoisy[kPose], Matrix(ΣOdo))
            drecSym = Symbol("deadreckon_x",kPose)
            addVariable!(fg, drecSym, Pose2, solvable=0)
            addFactor!(fg, [Symbol("x",lastSolveIdx); drecSym], drec, solvable=0)
            global drecPoses[kPose,:] = accumulateFactorMeans(fg, [Symbol("x",lastSolveIdx,"deadreckon_x",kPose,"f1")])
            @info "drecPoses x$(kPose) = $(drecPoses[kPose,:])"
            @info "drecμ x$(kPose) = $(drec.Zij.μ)"
        end

        # save
        #prefix = lpad(kPose,5,'0')
        #saveDFG(fg, "$(getLogPath(fg))/$(prefix)_fg")

    end
    #saveDFG(fg, "$(getLogPath(fg))/final_fg")
end
@info "time: $(tRun) s"
pl = plot(x=drecPoses[:,1], y=drecPoses[:,2], Geom.line, Guide.xlabel("x"), Guide.ylabel("y"))
```
It seems like I am making things more complicated than they need to be. Is it intended to add a new deadreckon factor and variable for every pose that i want to estimate using `accumulateFactorMeans` or is it possible to modify the factor in-graph? 

Is there an easy approach to parallelize the DRT and factor graph solve computations?

Best,
Leo

dehann, 2021-02-07:
Hi @lemauee ,

> Is there an easy approach to parallelize the DRT and factor graph solve computations?

Yes, if I understand your question right.  That is the idea with dead reckon tether (DRT).  The trick is something called `solvable=0` to "disengage" a variable or factor from regular solving.  
```julia
# to disengage from regular solving but still be useful in the graph
addVariable!(..., solvable=0) # default   =1
addFactor!(..., solvable=0)
```

You can then still do user, `@async` or likely `@spawn` operations on the graph as much you like regarding DRT or otherwise.

I updated the Caesar docs a bit to try help.  Perhaps look through that to see if that helps answer your question better:  JuliaRobotics/Caesar.jl#682 .

> or is it possible to modify the factor in-graph?

think you perhaps looking for `MutablePose2Pose2Gaussian`

Also see code example here from a while back (one a car another an auv).  In these we employed both "fixed lag" solving using the tree, as well as DRT.  Note that the solver was called by async process every 10 poses, and would allow the vehicle to accumulate an additional 10 poses while solving in the background.  Simultaneous to all that multiple DRTs would be updated while a bunch of other graph operations are underway. 

Start the solve manager:
https://github.com/JuliaRobotics/Caesar.jl/blob/master/examples/wheeled/racecar/ros/CarFeedMonoFlux.jl#L160

https://github.com/JuliaRobotics/Caesar.jl/blob/bd367355ae3ca995476ee0c00cbecba27d371807/examples/marine/auv/Sandshark/MsgHandlers.jl#L142

https://github.com/JuliaRobotics/Caesar.jl/blob/bd367355ae3ca995476ee0c00cbecba27d371807/examples/marine/auv/Sandshark/MsgHandlers.jl#L150-L155

https://github.com/JuliaRobotics/Caesar.jl/blob/bd367355ae3ca995476ee0c00cbecba27d371807/examples/marine/auv/Sandshark/MsgHandlers.jl#L174

Also just keeping @Affie in the loop who might be able to respond quicker.  I'm semi hiding to finish IIF 1010 :-) but please keep the comments and issues coming.  Any issue identified and/or resolved is good progress, thanks!

Best,
Dehann

lemauee, 2021-02-08:
Thanks for the quick reply!

The concept of using DRT and the regular solve was already known to me (great video on that :)), but the implementation of disengaging a variable using the `solvable` flag and then using the `@async` or `@spawn` macros was the implementation hint I needed. 

However, one question still persists: Is changing the value of the MutalbePose2Pose2Gaussian Factor still possible once its added to the graph? In my code above I currently add a new DRT factor for every value I want to obtain using accumulateFactorMeans. This is why I think I'm doing something wrong there.

```julia
# solve
        if mod(kPose,solveStep) == 1 # full solve
            if firstsolve
                tree, smt, hist = solveTree!(fg; multithread=true)
                firstsolve = false
            else
                tree, smt, hist = solveTree!(fg, tree; multithread=true)
            end
            drecSym = Symbol("deadreckon_x",kPose)
            addVariable!(fg, drecSym, Pose2, solvable=0)
            global drec = MutablePose2Pose2Gaussian(MvNormal(zeros(3), Matrix{Float64}(LinearAlgebra.I, 3,3)))               
            addFactor!(fg, [Symbol("x",kPose); drecSym], drec, solvable=0)
            global drecPoses[kPose,:] = accumulateFactorMeans(fg, [Symbol("x",drecIdx,"deadreckon_x",drecIdx,"f1")])
            lastSolveIdx = kPose
        else # DRT
            accumulateDiscreteLocalFrame!(drec, mTrajNoisy[kPose], Matrix(ΣOdo))
            drecSym = Symbol("deadreckon_x",kPose)
            addVariable!(fg, drecSym, Pose2, solvable=0)
            addFactor!(fg, [Symbol("x",lastSolveIdx); drecSym], drec, solvable=0)
            global drecPoses[kPose,:] = accumulateFactorMeans(fg, [Symbol("x",lastSolveIdx,"deadreckon_x",kPose,"f1")])
            @info "drecPoses x$(kPose) = $(drecPoses[kPose,:])"
            @info "drecμ x$(kPose) = $(drec.Zij.μ)"
        end
```

Semi hiding seems like a good approach to get things done ;)

Best,
Leo

lemauee, 2021-02-08:
Maybe to reformulate the question, these two approaches would make it unnecessary to add a new DRT factor every time:

Is there a way to call `accumulateDiscreteLocalFrame!` on a Factor in the graph (handing the fg and a symbol, not the factor itself)?

Or the other way round: can `accumulateFactorMeans` be called on the `MutablePose2Pose2Gaussian` without adding it to the fg (some prior would be required for that, but this could still be obtained from the fg).

Best,
Leo

dehann, 2021-02-09:
Hi Leo,

> Is there a way to call accumulateDiscreteLocalFrame! on a Factor in the graph (handing the fg and a symbol, not the factor itself)?

Note #402 and backport #403, I added the additional API on `accumulateDiscreteLocalFrame!`.

> can accumulateFactorMeans be called on the MutablePose2Pose2Gaussian without adding it to the fg

uhmmm, ahhh, hmm... we had not initially planned to use it that way but worth thinking about.  Part of why we avoided 'separate factor calculations' is there is quite a lot of plumbing happening under the hood, and separately instantiating the memory for single factor evaluations can be much more tedious.  

Could I perhaps ask about an alternative, would your case work if you created a small temporary factor graph that was not part of the other "main" fg?  I myself started using a pattern:

```julia
fg = .....

drt = MutablePose2Pose2(...)

# a separate temporary graph.
tfg = generateCanonicalFG_ZeroPose2()
addVariable!(tfg, :drt0, Pose2)
addFactor!(tfg, [:x0;:drt0], drt)

# use both fg and tfg
```

I know this might look like more work, however, the idea here would be to separate `fg` and `tfg` entirely.  If you want to minimize work and trust that graph management in `fg` is working well enough then `solvable=0` should do the trick.  I think we would very much like to know of cases where graph management is not working as expected or needed so that we can fix that.

In general, don't be scared to make many `fg*` objects, deepcopy, etc.  Thats how we'd recommend using the packages, i.e.:
```julia
fg = ...

fg2 = deepcopy(fg)

getSolverParams(fg2).async = true
solveTree!(fg2);

# keep going on others
addVariable!(fg, ...)

# do other stuff
plotSLAM2D(deepcopy(fg))

# etc.
```

Best,
Dehann

lemauee, 2021-02-09:
Thanks for the quick extension of the api, the first approach is the better of my two suggested ones, the second one was just an option if the first one would not be so easy to implement. The code for solving now looks reasonably clean:

```julia
 # solve
        if mod(kPose,solveStep) == 1 # full solve
            if firstsolve
                tree, smt, hist = solveTree!(fg; multithread=true)
                firstsolve = false
            else
                tree, smt, hist = solveTree!(fg, tree; multithread=true)
            end
            lastSolveIdx = kPose
            addVariable!(fg, Symbol("drt_x",kPose), Pose2, solvable=0)
            drec = MutablePose2Pose2Gaussian(MvNormal(zeros(3), Matrix{Float64}(LinearAlgebra.I, 3,3)))               
            addFactor!(fg, [Symbol("x",kPose); Symbol("drt_x",kPose)], drec, solvable=0)
        else # DRT
            accumulateDiscreteLocalFrame!(fg,Symbol("x",lastSolveIdx,"drt_x",lastSolveIdx,"f1"), mTrajNoisy[kPose], Matrix(ΣOdo))
        end
        global drecPoses[kPose,:] = accumulateFactorMeans(fg, [Symbol("x",lastSolveIdx,"drt_x",lastSolveIdx,"f1")])
        @info "drecPoses x$(kPose) = $(drecPoses[kPose,:])"
        @info "drecμ x$(kPose) = $(getFactor(fg, Symbol("x",lastSolveIdx,"drt_x",lastSolveIdx,"f1")).Zij.μ)"
```

Unfortunately I think i already tried to do what you now did as an alternative API now and I am stuck on the same error again:

```julia-repl
MethodError: no method matching accumulateDiscreteLocalFrame!(::DFGFactor{CommonConvWrapper{MutablePose2Pose2Gaussian},2}, ::Array{Float64,1}, ::Array{Float64,2}, ::Float64; Fk=[0.9999999292242013 -0.0003762334278278348 0.0; 0.0003762334278278348 0.9999999292242013 0.0; 0.0 0.0 1.0], Gk=[1.0 0.0 0.0; 0.0 1.0 0.0; 0.0 0.0 1.0], Phik=[0.9999999292242013 0.0003762334278278348 -0.0069543805839192745; -0.0003762334278278348 0.9999999292242013 -0.004441156065982448; 0.0 0.0 1.0])
```

accumulateDiscreteLocalFrame expects a "pure" MutablePose2Pose2Gaussian, not the wrapped version it gets now.

Another problem I had was that the backport did not make it to the registered version by now, i pasted the code into my document to circumvent this as i did not want to change to the dev branch yet. Is this normal and I should just wait some time or is something messed up with my environment? (the last backport on the Pose2Point2 worked fine for me)

Just for completeness, the whole example changed to the new api so you can try If it works on your machine:
```julia
## Read Gt Input from files

using CSV

associationLm = CSV.File("association_lm.txt", header=["iPose", "jLm"])
gtLm = CSV.File("gt_lm.txt", header=["x", "y", "θ"])
gtTraj = CSV.File("gt_traj.txt", header=["x", "y", "θ"])

## Generate bearing range measurements from groundtruth

using CoordinateTransformations, TransformUtils, LinearAlgebra

Tinv(x,y,θ) = inv(Translation(x,y) ∘ LinearMap(R(θ)))

mTrajGt = [Vector{Float64}(undef,3) for _ = 1:length(gtTraj)]
mLmGt = [Dict{Int,Vector{Float64}}() for _ = 1:length(gtTraj)]

for kPose = 1:length(gtTraj)

    # odometry
    if kPose > 1 # no odemetry for first pose, prior instead
        trans = Tinv(gtTraj[kPose-1].x,gtTraj[kPose-1].y,gtTraj[kPose-1].θ)([gtTraj[kPose].x, gtTraj[kPose].y])
        rot = TransformUtils.wrapRad(gtTraj[kPose].θ - gtTraj[kPose-1].θ)
        mTrajGt[kPose] = vcat(trans,rot)
    end

    # landmark measurements
    lmIdx = associationLm.jLm[findall(associationLm.iPose .== kPose)]
    for kLm = lmIdx
        trans = Tinv(gtTraj[kPose].x,gtTraj[kPose].y,gtTraj[kPose].θ)([gtLm[kLm].x, gtLm[kLm].y])
        rot = TransformUtils.wrapRad(gtLm[kLm].θ - gtTraj[kPose].θ)
        mLmGt[kPose][kLm] = vcat(trans,rot)
    end
end

[... 89 lines cut ...]
            drec = MutablePose2Pose2Gaussian(MvNormal(zeros(3), Matrix{Float64}(LinearAlgebra.I, 3,3)))               
            addFactor!(fg, [Symbol("x",kPose); Symbol("drt_x",kPose)], drec, solvable=0)
        else # DRT
            accumulateDiscreteLocalFrame!(fg,Symbol("x",lastSolveIdx,"drt_x",lastSolveIdx,"f1"), mTrajNoisy[kPose], Matrix(ΣOdo))
        end
        global drecPoses[kPose,:] = accumulateFactorMeans(fg, [Symbol("x",lastSolveIdx,"drt_x",lastSolveIdx,"f1")])
        @info "drecPoses x$(kPose) = $(drecPoses[kPose,:])"
        @info "drecμ x$(kPose) = $(getFactor(fg, Symbol("x",lastSolveIdx,"drt_x",lastSolveIdx,"f1")).Zij.μ)"

        # save
        #prefix = lpad(kPose,5,'0')
        #saveDFG(fg, "$(getLogPath(fg))/$(prefix)_fg")

    end
    #saveDFG(fg, "$(getLogPath(fg))/final_fg")
end
@info "time: $(tRun) s"
pl = plot(x=drecPoses[:,1], y=drecPoses[:,2], Geom.line, Guide.xlabel("x"), Guide.ylabel("y"))
```
[association_lm.txt](https://github.com/JuliaRobotics/RoME.jl/files/5950322/association_lm.txt)
[gt_lm.txt](https://github.com/JuliaRobotics/RoME.jl/files/5950324/gt_lm.txt)
[gt_traj.txt](https://github.com/JuliaRobotics/RoME.jl/files/5950325/gt_traj.txt)

Best,
Leo

Affie, 2021-02-09:
> Another problem I had was that the backport did not make it to the registered version by now, i pasted the code into my document to circumvent this as i did not want to change to the dev branch yet. [...]

It's not your environment, it was not released yet, you can use `pkg> add RoME#release/v0.12` if you want to check it out in the meantime.

lemauee, 2021-02-09:
Thanks @Affie for the hint, the difference between "tagged as release" and "registered" versions is something I'll be aware of from now on. Will staying on the release branch be as instable as working with the current master? Or is staying on the release branch something you could recommend generally to not miss these important updates?

As for the error, the problem (unsuprisingly) is still the same: `accumulateDiscreteLocalFrame!` expects a "pure" MutablePose2Pose2Gaussian, not the wrapped version it gets now

Affie, 2021-02-09:
The release branches are very stable, we normally register a release on them right away so it should be fine to stay on the registered versions.  
Let me see if the error is a quick fix or not. I haven't really worked with the tether before.

Affie, 2021-02-09:
Hi @lemauee, Please give the `release/v0.12` branch another go and let me know if it works now. I haven't fully tested it myself. 
I can register it for you if it works.

lemauee, 2021-02-09:
Hi @Affie, it now works as expected, I also changed my info line on the current mean of the active drt according to your fix:
```julia
@info "drecμ x$(kPose) = $(RoME.getFactorFunction(fg, Symbol("x",lastSolveIdx,"drt_x",lastSolveIdx,"f1")).Zij.μ)"
```
for anyone who wants to give it a try, here's the complete code again (change endPose and solveStep according to how long you want to occupy your machine ;), dataset finishes at pose 375), files are provided in previous post :
```julia
## Read Gt Input from files

using CSV

associationLm = CSV.File("association_lm.txt", header=["iPose", "jLm"])
gtLm = CSV.File("gt_lm.txt", header=["x", "y", "θ"])
gtTraj = CSV.File("gt_traj.txt", header=["x", "y", "θ"])

## Generate bearing range measurements from groundtruth

using CoordinateTransformations, TransformUtils, LinearAlgebra

Tinv(x,y,θ) = inv(Translation(x,y) ∘ LinearMap(R(θ)))

mTrajGt = [Vector{Float64}(undef,3) for _ = 1:length(gtTraj)]
mLmGt = [Dict{Int,Vector{Float64}}() for _ = 1:length(gtTraj)]

for kPose = 1:length(gtTraj)

    # odometry
    if kPose > 1 # no odemetry for first pose, prior instead
        trans = Tinv(gtTraj[kPose-1].x,gtTraj[kPose-1].y,gtTraj[kPose-1].θ)([gtTraj[kPose].x, gtTraj[kPose].y])
        rot = TransformUtils.wrapRad(gtTraj[kPose].θ - gtTraj[kPose-1].θ)
        mTrajGt[kPose] = vcat(trans,rot)
    end

    # landmark measurements
    lmIdx = associationLm.jLm[findall(associationLm.iPose .== kPose)]
    for kLm = lmIdx
        trans = Tinv(gtTraj[kPose].x,gtTraj[kPose].y,gtTraj[kPose].θ)([gtLm[kLm].x, gtLm[kLm].y])
        rot = TransformUtils.wrapRad(gtLm[kLm].θ - gtTraj[kPose].θ)
        mLmGt[kPose][kLm] = vcat(trans,rot)
    end
end

## Add noise to ground truth measurements

using Distributions, Random

Random.seed!(42); # reproducibility

ΣOdo = Diagonal([0.01,0.01,0.5*pi/180].^2)
ΣLm = Diagonal([0.01,0.01,0.5*pi/180].^2)
# ΣOdo = Diagonal([0.1,0.1,5*pi/180].^2)
# ΣLm = Diagonal([0.1,0.1,5*pi/180].^2)

noiseOdo = MvNormal([0,0,0],ΣOdo)
noiseLm = MvNormal([0,0,0],ΣLm)

mTrajNoisy = [Vector{Float64}(undef,3) for _ = 1:length(gtTraj)]
mLmNoisy = [Dict{Int,Vector{Float64}}() for _ = 1:length(gtTraj)]

for kPose = 1:length(gtTraj)

    # odometry
    if kPose > 1 # no odemetry for first pose, prior instead
        mTrajNoisy[kPose] = mTrajGt[kPose] + vec(rand(noiseOdo,1))
    end

    # landmark measurements
    for kLm = keys(mLmGt[kPose])
        mLmNoisy[kPose][kLm] = mLmGt[kPose][kLm] + vec(rand(noiseLm,1))
    end
end

## Build fg from measurements and solve incrementally

using Distributed, RoME
@everywhere using RoME

tRun = @elapsed begin
    global endPose = 10;
    global solveStep = 5;
    global fg = LightDFG{SolverParams}(solverParams=SolverParams(logpath="speed"))
    getSolverParams(fg).useMsgLikelihoods = false
    # getSolverParams(fg).isfixedlag = true
    # getSolverParams(fg).qfl = 50
    # getSolverParams(fg).limitfixeddown = true
    firstsolve = true
    drecPoses = Array{Float64,2}(undef,endPose,3)

    function addVarIfMissing!(fg,symbol,type,tags)
        if (symbol in ls(fg)) == false
            @info "adding missing variable $(symbol)"
            addVariable!(fg, symbol, type, tags = tags)
        end
    end

    for kPose = 1:endPose
        
        global fg,tree,smt,hist,lastSolveIdx,firstsolve

        # odometry
        if kPose > 1 # no odometry for first pose
            fromPose = Symbol("x", kPose-1)
            toPose = Symbol("x", kPose)
            addVarIfMissing!(fg, fromPose, Pose2, [:POSE])
            addVarIfMissing!(fg, toPose, Pose2, [:POSE])   
            addFactor!(fg, [fromPose, toPose], Pose2Pose2(MvNormal(mTrajNoisy[kPose], ΣOdo)), graphinit = false)
            initManual!(fg, toPose, [Symbol("x",kPose-1,"x",kPose,"f",1)]) # doing this because I initlialize the same way in GTSAM
        else # prior instead
            addVariable!(fg, :x1, Pose2)
            addFactor!(fg, [:x1], PriorPose2(MvNormal([gtTraj[1].x, gtTraj[1].y, gtTraj[1].θ], Diagonal([0.01,0.01,0.5*pi/180].^2))))
        end

        # landmark measurements
        fromPose = Symbol("x", kPose)
        for kLm = collect(keys(mLmGt[kPose]))
            toLm = Symbol("l", kLm)
            addVarIfMissing!(fg, fromPose, Pose2, [:POSE])
            addVarIfMissing!(fg, toLm, Pose2, [:LANDMARK])
            addFactor!(fg, [fromPose, toLm], Pose2Pose2(MvNormal(mLmNoisy[kPose][kLm], ΣLm)))
        end

        # solve
        if mod(kPose,solveStep) == 1 # full solve
            if firstsolve
                tree, smt, hist = solveTree!(fg; multithread=true)
                firstsolve = false
            else
                tree, smt, hist = solveTree!(fg, tree; multithread=true)
            end
            lastSolveIdx = kPose
            addVariable!(fg, Symbol("drt_x",kPose), Pose2, solvable=0)
            drec = MutablePose2Pose2Gaussian(MvNormal(zeros(3), Matrix{Float64}(LinearAlgebra.I, 3,3)))               
            addFactor!(fg, [Symbol("x",kPose); Symbol("drt_x",kPose)], drec, solvable=0)
        else # DRT
            accumulateDiscreteLocalFrame!(fg,Symbol("x",lastSolveIdx,"drt_x",lastSolveIdx,"f1"), mTrajNoisy[kPose], Matrix(ΣOdo))
        end
        global drecPoses[kPose,:] = accumulateFactorMeans(fg, [Symbol("x",lastSolveIdx,"drt_x",lastSolveIdx,"f1")])
        @info "drecPoses x$(kPose) = $(drecPoses[kPose,:])"
        @info "drecμ x$(kPose) = $(RoME.getFactorFunction(fg, Symbol("x",lastSolveIdx,"drt_x",lastSolveIdx,"f1")).Zij.μ)"

        # save
        #prefix = lpad(kPose,5,'0')
        #saveDFG(fg, "$(getLogPath(fg))/$(prefix)_fg")

    end
    #saveDFG(fg, "$(getLogPath(fg))/final_fg")
end
@info "time: $(tRun) s"

## plot

using Gadfly
pl = plot(x=drecPoses[:,1], y=drecPoses[:,2], Geom.line, Guide.xlabel("x"), Guide.ylabel("y"))
```

Affie, 2021-02-09:
v0.12.2 should be in the registry soon: see https://github.com/JuliaRobotics/RoME.jl/issues/406

lemauee, 2021-02-09:
The results for the 150 pose example now look as expected:

[image]

Remark: `Geom.line` had to be changed to `Geom.path` in the code above or the plot gets messed up.

From my side, the issue can be closed now :)

Best, Leo

dehann, 2021-02-09:
Great, thanks!

closed 2021-02-09 by dehann