JuliaRobotics/RoME.jl #407: Error running example using multiple processes

lemauee, 2021-02-09:
Hi, I just wanted to run the example from #396 using multiple processes and got the following error:

```julia-repl
ERROR: LoadError: TaskFailedException:
RemoteException(2, CapturedException(KeyError(TimeZones [f269a46b-ccf7-5d73-abea-4c690281aa53]), Any[(getindex at dict.jl:467 [inlined], 1), (root_module at loading.jl:968 [inlined], 1), (deserialize_module at Serialization.jl:953, 1), (handle_deserialize at Serialization.jl:855, 1), (deserialize at Serialization.jl:773, 1), (deserialize_datatype at Serialization.jl:1251, 1), (handle_deserialize at Serialization.jl:826, 1), (deserialize at Serialization.jl:773, 1), (handle_deserialize at Serialization.jl:833, 1), (deserialize at Serialization.jl:1350, 1), (handle_deserialize at Serialization.jl:837, 1), (deserialize at Serialization.jl:773 [inlined], 1), (deserialize_dict at Serialization.jl:1366, 1), (deserialize at Serialization.jl:1373, 1), (handle_deserialize at Serialization.jl:842, 1), (deserialize at Serialization.jl:1350, 1), (handle_deserialize at Serialization.jl:837, 1), (deserialize at Serialization.jl:1338, 1), (handle_deserialize at Serialization.jl:842, 1), (deserialize at Serialization.jl:773, 1), (#5 at Serialization.jl:932, 1), (ntuple at ntuple.jl:18, 1), (deserialize_tuple at Serialization.jl:932, 1), (handle_deserialize at Serialization.jl:816, 1), (deserialize at Serialization.jl:773 [inlined], 1), (deserialize_msg at messages.jl:99, 1), (#invokelatest#1 at essentials.jl:710 [inlined], 1), (invokelatest at essentials.jl:709 [inlined], 1), (message_handler_loop at process_messages.jl:185, 1), (process_tcp_streams at process_messages.jl:142, 1), (#99 at task.jl:356, 1)]))
Stacktrace:
 [1] error(::RemoteException) at ./error.jl:42
 [2] approxCliqMarginalUp!(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}, ::Array{LikelihoodMessage,1}; N::Int64, dbg::Bool, multiproc::Bool, logger::Base.CoreLogging.SimpleLogger, iters::Int64, drawpdf::Bool) at /home/leopold/.julia/packages/IncrementalInference/ShzPg/src/CliqStateMachineUtils.jl:379
 [3] __doCliqUpSolveInitialized!(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}) at /home/leopold/.julia/packages/IncrementalInference/ShzPg/src/CliqStateMachineUtils.jl:30
 [4] solveUp_StateMachine(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}) at /home/leopold/.julia/packages/IncrementalInference/ShzPg/src/CliqueStateMachine.jl:333
 [5] (::StateMachine{CliqStateMachineContainer})(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}, ::Nothing; pollinterval::Float64, breakafter::Function, verbose::Bool, verbosefid::Base.TTY, verboseXtra::IncrementalInference.CliqStatus, iterlimit::Int64, injectDelayBefore::Nothing, recordhistory::Bool, housekeeping_cb::IncrementalInference.var"#377#379"{IncrementalInference.TreeClique}) at /home/leopold/.julia/packages/FunctionalStateMachine/2JZFG/src/StateMachine.jl:82
 [6] initStartCliqStateMachine!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree, ::IncrementalInference.TreeClique, ::Nothing; oldcliqdata::BayesTreeNodeData, verbose::Bool, verbosefid::Base.TTY, drawtree::Bool, show::Bool, incremental::Bool, limititers::Int64, upsolve::Bool, downsolve::Bool, recordhistory::Bool, delay::Bool, logger::Base.CoreLogging.SimpleLogger, solve_progressbar::ProgressMeter.ProgressUnknown, algorithm::Symbol) at /home/leopold/.julia/packages/IncrementalInference/ShzPg/src/CliqueStateMachine.jl:63
 [7] tryCliqStateMachineSolve!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree, ::Int64, ::Nothing; oldtree::MetaBayesTree, verbose::Bool, verbosefid::Base.TTY, drawtree::Bool, limititers::Int64, downsolve::Bool, incremental::Bool, delaycliqs::Array{Symbol,1}, recordcliqs::Array{Symbol,1}, solve_progressbar::ProgressMeter.ProgressUnknown, algorithm::Symbol) at /home/leopold/.julia/packages/IncrementalInference/ShzPg/src/SolverAPI.jl:110
 [8] (::IncrementalInference.var"#432#435"{MetaBayesTree,Bool,Bool,Base.TTY,Int64,Bool,Bool,Array{Symbol,1},Array{Symbol,1},Symbol,LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree,Nothing,ProgressMeter.ProgressUnknown,Int64})() at ./threadingconstructs.jl:169
Stacktrace:
 [1] [ Info: monitorCSMs: all tasks done
sync_end(::Channel{Any}) at ./task.jl:314
 [2] macro expansion at ./task.jl:333 [inlined]
 [3] taskSolveTree!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree, ::Nothing; oldtree::MetaBayesTree, drawtree::Bool, verbose::Bool, verbosefid::Base.TTY, limititers::Int64, limititercliqs::Array{Pair{Symbol,Int64},1}, downsolve::Bool, incremental::Bool, multithread::Bool, skipcliqids::Array{Symbol,1}, recordcliqs::Array{Symbol,1}, delaycliqs::Array{Symbol,1}, smtasks::Array{Task,1}, algorithm::Symbol) at /home/leopold/.julia/packages/IncrementalInference/ShzPg/src/SolverAPI.jl:49
 [4] solveTree!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree; timeout::Nothing, storeOld::Bool, verbose::Bool, verbosefid::Base.TTY, delaycliqs::Array{Symbol,1}, recordcliqs::Array{Symbol,1}, limititercliqs::Array{Pair{Symbol,Int64},1}, injectDelayBefore::Nothing, skipcliqids::Array{Symbol,1}, eliminationOrder::Nothing, variableOrder::Nothing, eliminationConstraints::Array{Symbol,1}, variableConstraints::Nothing, smtasks::Array{Task,1}, dotreedraw::Array{Int64,1}, runtaskmonitor::Bool, algorithm::Symbol, multithread::Bool) at /home/leopold/.julia/packages/IncrementalInference/ShzPg/src/SolverAPI.jl:371
 [5] top-level scope at /home/leopold/Nextcloud/Studium/Masterarbeit/svn/julia/mmiSAM/examples/speed/speed.jl:118
 [6] top-level scope at timing.jl:233
 [7] include_string(::Function, ::Module, ::String, ::String) at ./loading.jl:1088
 [8] include_string(::Module, ::String, ::String) at ./loading.jl:1096
 [9] invokelatest(::Any, ::Any, ::Vararg{Any,N} where N; kwargs::Base.Iterators.Pairs{Union{},Union{},Tuple{},NamedTuple{(),Tuple{}}}) at ./essentials.jl:710
 [10] invokelatest(::Any, ::Any, ::Vararg{Any,N} where N) at ./essentials.jl:709
 [11] inlineeval(::Module, ::String, ::Int64, ::Int64, ::String; softscope::Bool) at /home/leopold/.vscode-oss/extensions/julialang.language-julia-1.0.10/scripts/packages/VSCodeServer/src/eval.jl:185
 [12] (::VSCodeServer.var"#61#65"{String,Int64,Int64,String,Module,Bool,VSCodeServer.ReplRunCodeRequestParams})() at /home/leopold/.vscode-oss/extensions/julialang.language-julia-1.0.10/scripts/packages/VSCodeServer/src/eval.jl:144
 [13] withpath(::VSCodeServer.var"#61#65"{String,Int64,Int64,String,Module,Bool,VSCodeServer.ReplRunCodeRequestParams}, ::String) at /home/leopold/.vscode-oss/extensions/julialang.language-julia-1.0.10/scripts/packages/VSCodeServer/src/repl.jl:124
 [14] (::VSCodeServer.var"#60#64"{String,Int64,Int64,String,Module,Bool,Bool,VSCodeServer.ReplRunCodeRequestParams})() at /home/leopold/.vscode-oss/extensions/julialang.language-julia-1.0.10/scripts/packages/VSCodeServer/src/eval.jl:142
 [15] hideprompt(::VSCodeServer.var"#60#64"{String,Int64,Int64,String,Module,Bool,Bool,VSCodeServer.ReplRunCodeRequestParams}) at /home/leopold/.vscode-oss/extensions/julialang.language-julia-1.0.10/scripts/packages/VSCodeServer/src/repl.jl:36
 [16] (::VSCodeServer.var"#59#63"{String,Int64,Int64,String,Module,Bool,Bool,VSCodeServer.ReplRunCodeRequestParams})() at /home/leopold/.vscode-oss/extensions/julialang.language-julia-1.0.10/scripts/packages/VSCodeServer/src/eval.jl:110
 [17] with_logstate(::Function, ::Any) at ./logging.jl:408
 [18] with_logger at ./logging.jl:514 [inlined]
 [19] (::VSCodeServer.var"#58#62"{VSCodeServer.ReplRunCodeRequestParams})() at /home/leopold/.vscode-oss/extensions/julialang.language-julia-1.0.10/scripts/packages/VSCodeServer/src/eval.jl:109
 [20] #invokelatest#1 at ./essentials.jl:710 [inlined]
 [21] invokelatest(::Any) at ./essentials.jl:709
 [22] macro expansion at /home/leopold/.vscode-oss/extensions/julialang.language-julia-1.0.10/scripts/packages/VSCodeServer/src/eval.jl:27 [inlined]
 [23] (::VSCodeServer.var"#56#57")() at ./task.jl:356
in expression starting at /home/leopold/Nextcloud/Studium/Masterarbeit/svn/julia/mmiSAM/examples/speed/speed.jl:
```

The only thing I did in the REPL before executing the Code from #396 was

```julia
using Distributed
addprocs(3)
```

on single process it runs fine

Best,
Leo

dehann, 2021-02-09:
Hi Leo,

Oops that looks unfamiliar, assuming you doing something similar to this -- I found the following sequence is the most reliable (without using Revise.jl):
```julia
using Distributed
using RoME
addprocs(3)
using RoME
@everywhere RoME 

# some local precompiling to speed up later usage
warmUpSolverJIT()
```

Would you mind quickly trying this sequence on that example and let us know?  If that does not do it and something new is happening, the fastest place to debug looks like area of the code called CSM (`CliqStateMachine`).

Best,
Dehann

lemauee, 2021-02-10:
Hi Dehann,

First off I found a typo, I assume you meant `@everywhere using RoME` and not `@everywhere RoME` :)

Now to the long story for the error, I'll post a tl;dr version below:

VSCode uses Revise.jl by default (which Juno did not do by default), when running `@everywhere using RoME` it already has compilation errors I did not put much interest in until now:
```julia-repl
julia> @everywhere using RoME
      From worker 3:    [ Info: Multithreaded  convolutions possible, Threads.nthreads()=1.  See `addFactor!(.;threadmodel=MultiThreaded)`.
      From worker 4:    [ Info: Multithreaded  convolutions possible, Threads.nthreads()=1.  See `addFactor!(.;threadmodel=MultiThreaded)`.
      From worker 2:    [ Info: Multithreaded  convolutions possible, Threads.nthreads()=1.  See `addFactor!(.;threadmodel=MultiThreaded)`.
      From worker 3:    WARNING: Method definition dot(Any, Any, Any) in module LinearAlgebra at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/LinearAlgebra/src/generic.jl:924 overwritten in module Optim at /home/leopold/.julia/packages/Optim/Agd3B/src/multivariate/precon.jl:23.
      From worker 3:      ** incremental compilation may be fatally broken for this module **
      From worker 3:
      From worker 4:    WARNING: Method definition dot(Any, Any, Any) in module LinearAlgebra at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/LinearAlgebra/src/generic.jl:924 overwritten in module Optim at /home/leopold/.julia/packages/Optim/Agd3B/src/multivariate/precon.jl:23.
      From worker 4:      ** incremental compilation may be fatally broken for this module **
      From worker 4:
      From worker 2:    WARNING: Method definition dot(Any, Any, Any) in module LinearAlgebra at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/LinearAlgebra/src/generic.jl:924 overwritten in module Optim at /home/leopold/.julia/packages/Optim/Agd3B/src/multivariate/precon.jl:23.
      From worker 2:      ** incremental compilation may be fatally broken for this module **
      From worker 2:
      From worker 3:    [ Info: KernelDensityEstimate.FORCE_EVAL_DIRECT = true
      From worker 4:    [ Info: KernelDensityEstimate.FORCE_EVAL_DIRECT = true
      From worker 2:    [ Info: KernelDensityEstimate.FORCE_EVAL_DIRECT = true
      From worker 3:    ┌ Warning: The call to compilecache failed to create a usable precompiled cache file for IncrementalInference [904591bb-b899-562f-9e6f-b8df64c7d480]
      From worker 3:    │   exception = Required dependency Optim [429524aa-4258-5aef-a3af-852621145aeb] failed to load from a cache file.
      From worker 3:    └ @ Base loading.jl:1042
      From worker 4:    ┌ Warning: The call to compilecache failed to create a usable precompiled cache file for IncrementalInference [904591bb-b899-562f-9e6f-b8df64c7d480]
      From worker 4:    │   exception = Required dependency Optim [429524aa-4258-5aef-a3af-852621145aeb] failed to load from a cache file.
      From worker 4:    └ @ Base loading.jl:1042
      From worker 2:    ┌ Warning: The call to compilecache failed to create a usable precompiled cache file for IncrementalInference [904591bb-b899-562f-9e6f-b8df64c7d480]
      From worker 2:    │   exception = Required dependency Optim [429524aa-4258-5aef-a3af-852621145aeb] failed to load from a cache file.
      From worker 2:    └ @ Base loading.jl:1042
      From worker 4:    [ Info: Multithreaded  convolutions possible, Threads.nthreads()=1.  See `addFactor!(.;threadmodel=MultiThreaded)`.
      From worker 3:    [ Info: Multithreaded  convolutions possible, Threads.nthreads()=1.  See `addFactor!(.;threadmodel=MultiThreaded)`.
      From worker 2:    [ Info: Multithreaded  convolutions possible, Threads.nthreads()=1.  See `addFactor!(.;threadmodel=MultiThreaded)`.
      From worker 3:    WARNING: Method definition dot(Any, Any, Any) in module LinearAlgebra at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/LinearAlgebra/src/generic.jl:924 overwritten in module Optim at /home/leopold/.julia/packages/Optim/Agd3B/src/multivariate/precon.jl:23.
      From worker 3:      ** incremental compilation may be fatally broken for this module **
      From worker 3:
      From worker 4:    WARNING: Method definition dot(Any, Any, Any) in module LinearAlgebra at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/LinearAlgebra/src/generic.jl:924 overwritten in module Optim at /home/leopold/.julia/packages/Optim/Agd3B/src/multivariate/precon.jl:23.
      From worker 4:      ** incremental compilation may be fatally broken for this module **
      From worker 4:
      From worker 2:    WARNING: Method definition dot(Any, Any, Any) in module LinearAlgebra at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/LinearAlgebra/src/generic.jl:924 overwritten in module Optim at /home/leopold/.julia/packages/Optim/Agd3B/src/multivariate/precon.jl:23.
      From worker 2:      ** incremental compilation may be fatally broken for this module **
      From worker 2:
      From worker 3:    [ Info: KernelDensityEstimate.FORCE_EVAL_DIRECT = true
      From worker 4:    [ Info: KernelDensityEstimate.FORCE_EVAL_DIRECT = true
      From worker 2:    [ Info: KernelDensityEstimate.FORCE_EVAL_DIRECT = true
      From worker 4:    WARNING: ignoring conflicting import of DistributedFactorGraphs.compare into RoME
[... cut ...]
 [7] warmUpSolverJIT() at /home/leopold/.julia/packages/RoME/W4sOn/src/CanonicalGraphs.jl:226
 [8] top-level scope at REPL[10]:1

julia> [ Info: monitorCSMs: all tasks done
```
Can you reproduce this error? Instantiating a new environment, adding RoME and doing your prep-steps is a fast thing to check :) I am working on the version registered yesterday by @Affie ,  but the error is likely to  have been introduced earlier (I did not use multiprocess on the up-to-date version for a while). Sorry for the long post ;)

Best, Leo

lemauee, 2021-02-10:
tl;dr: Steps to reproduce error
Start julia in a an empty folder
```julia
(@v1.5) pkg> activate .
(julia_sandbox_2) pkg> add RoME
julia> using Distributed
julia> using RoME
julia> addprocs(3)
julia> using RoME
julia> @everywhere using RoME
julia> warmUpSolverJIT()
```
Things should crash during solve.

Affie, 2021-02-10:
Hi @lemauee, I think there should be an error logged in you "speed" folder from (logpath="speed"). Please check it and post the relevant section. 
I just saw the post above come through. I'll look in more detail...

lemauee, 2021-02-10:
@Affie contents of cliq1_stacktrace.txt:
```julia-repl
RemoteException(2, CapturedException(KeyError(TimeZones [f269a46b-ccf7-5d73-abea-4c690281aa53]), Any[(getindex at dict.jl:467 [inlined], 1), (root_module at loading.jl:968 [inlined], 1), (deserialize_module at Serialization.jl:953, 1), (handle_deserialize at Serialization.jl:855, 1), (deserialize at Serialization.jl:773, 1), (deserialize_datatype at Serialization.jl:1251, 1), (handle_deserialize at Serialization.jl:826, 1), (deserialize at Serialization.jl:773, 1), (handle_deserialize at Serialization.jl:833, 1), (deserialize at Serialization.jl:1350, 1), (handle_deserialize at Serialization.jl:837, 1), (deserialize at Serialization.jl:773 [inlined], 1), (deserialize_dict at Serialization.jl:1366, 1), (deserialize at Serialization.jl:1373, 1), (handle_deserialize at Serialization.jl:842, 1), (deserialize at Serialization.jl:1350, 1), (handle_deserialize at Serialization.jl:837, 1), (deserialize at Serialization.jl:1338, 1), (handle_deserialize at Serialization.jl:842, 1), (deserialize at Serialization.jl:773, 1), (#5 at Serialization.jl:932, 1), (ntuple at ntuple.jl:18, 1), (deserialize_tuple at Serialization.jl:932, 1), (handle_deserialize at Serialization.jl:816, 1), (deserialize at Serialization.jl:773 [inlined], 1), (deserialize_msg at messages.jl:99, 1), (#invokelatest#1 at essentials.jl:710 [inlined], 1), (invokelatest at essentials.jl:709 [inlined], 1), (message_handler_loop at process_messages.jl:185, 1), (process_tcp_streams at process_messages.jl:142, 1), (#99 at task.jl:356, 1)]))
Stacktrace:
 [1] error(::RemoteException) at ./error.jl:42
 [2] approxCliqMarginalUp!(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}, ::Array{LikelihoodMessage,1}; N::Int64, dbg::Bool, multiproc::Bool, logger::Base.CoreLogging.SimpleLogger, iters::Int64, drawpdf::Bool) at /home/leopold/.julia/packages/IncrementalInference/ShzPg/src/CliqStateMachineUtils.jl:379
 [3] __doCliqUpSolveInitialized!(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}) at /home/leopold/.julia/packages/IncrementalInference/ShzPg/src/CliqStateMachineUtils.jl:30
 [4] solveUp_StateMachine(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}) at /home/leopold/.julia/packages/IncrementalInference/ShzPg/src/CliqueStateMachine.jl:333
 [5] (::StateMachine{CliqStateMachineContainer})(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}, ::Nothing; pollinterval::Float64, breakafter::Function, verbose::Bool, verbosefid::Base.TTY, verboseXtra::IncrementalInference.CliqStatus, iterlimit::Int64, injectDelayBefore::Nothing, recordhistory::Bool, housekeeping_cb::IncrementalInference.var"#377#379"{IncrementalInference.TreeClique}) at /home/leopold/.julia/packages/FunctionalStateMachine/2JZFG/src/StateMachine.jl:82
 [6] initStartCliqStateMachine!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree, ::IncrementalInference.TreeClique, ::Nothing; oldcliqdata::BayesTreeNodeData, verbose::Bool, verbosefid::Base.TTY, drawtree::Bool, show::Bool, incremental::Bool, limititers::Int64, upsolve::Bool, downsolve::Bool, recordhistory::Bool, delay::Bool, logger::Base.CoreLogging.SimpleLogger, solve_progressbar::ProgressMeter.ProgressUnknown, algorithm::Symbol) at /home/leopold/.julia/packages/IncrementalInference/ShzPg/src/CliqueStateMachine.jl:63
 [7] tryCliqStateMachineSolve!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree, ::Int64, ::Nothing; oldtree::MetaBayesTree, verbose::Bool, verbosefid::Base.TTY, drawtree::Bool, limititers::Int64, downsolve::Bool, incremental::Bool, delaycliqs::Array{Symbol,1}, recordcliqs::Array{Symbol,1}, solve_progressbar::ProgressMeter.ProgressUnknown, algorithm::Symbol) at /home/leopold/.julia/packages/IncrementalInference/ShzPg/src/SolverAPI.jl:110
 [8] (::IncrementalInference.var"#432#435"{MetaBayesTree,Bool,Bool,Base.TTY,Int64,Bool,Bool,Array{Symbol,1},Array{Symbol,1},Symbol,LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree,Nothing,ProgressMeter.ProgressUnknown,Int64})() at ./threadingconstructs.jl:169
```

Affie, 2021-02-10:
I'm not getting the error on the master branches I'm currently on. I'll try in a fresh environment.

That stack trace just points to the place the error is rethrown. Are there any other error messages?

lemauee, 2021-02-10:
I just switched to the master branch in my test-env and still get the same error running dehanns "prep-steps"

Affie, 2021-02-10:
It may have to do with one of the dependencies upgradings. I see the original error message is in the first line of your previous post, but it's hard to decipher. It looks like it has something to do with TimeZones.jl or Serialization.jl

lemauee, 2021-02-10:
@Affie: The whole log folder and the error message it crashes with of my example
[speed.zip](https://github.com/JuliaRobotics/RoME.jl/files/5956748/speed.zip)
```julia
[ Info: Bayes Tree Complete
[ Info: Do tree based init-inference on tree
Solve Progress: approx max 24, at iter 4         Time: 0:00:03
RemoteException(2, CapturedException(KeyError(TimeZones [f269a46b-ccf7-5d73-abea-4c690281aa53]), Any[(getindex at dict.jl:467 [inlined], 1), (root_module at loading.jl:968 [inlined], 1), (deserialize_module at Serialization.jl:953, 1), (handle_deserialize at Serialization.jl:855, 1), (deserialize at Serialization.jl:773, 1), (deserialize_datatype at Serialization.jl:1251, 1), (handle_deserialize at Serialization.jl:826, 1), (deserialize at Serialization.jl:773, 1), (handle_deserialize at Serialization.jl:833, 1), (deserialize at Serialization.jl:1350, 1), (handle_deserialize at Serialization.jl:837, 1), (deserialize at Serialization.jl:773 [inlined], 1), (deserialize_dict at Serialization.jl:1366, 1), (deserialize at Serialization.jl:1373, 1), (handle_deserialize at Serialization.jl:842, 1), (deserialize at Serialization.jl:1350, 1), (handle_deserialize at Serialization.jl:837, 1), (deserialize at Serialization.jl:1338, 1), (handle_deserialize at Serialization.jl:842, 1), (deserialize at Serialization.jl:773, 1), (#5 at Serialization.jl:932, 1), (ntuple at ntuple.jl:18, 1), (deserialize_tuple at Serialization.jl:932, 1), (handle_deserialize at Serialization.jl:816, 1), (deserialize at Serialization.jl:773 [inlined], 1), (deserialize_msg at messages.jl:99, 1), (#invokelatest#1 at essentials.jl:710 [inlined], 1), (invokelatest at essentials.jl:709 [inlined], 1), (message_handler_loop at process_messages.jl:185, 1), (process_tcp_streams at process_messages.jl:142, 1), (#99 at task.jl:356, 1)]))
Stacktrace:
 [1] error(::RemoteException) at ./error.jl:42
 [2] approxCliqMarginalUp!(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}, ::Array{LikelihoodMessage,1}; N::Int64, dbg::Bool, multiproc::Bool, logger::Base.CoreLogging.SimpleLogger, iters::Int64, drawpdf::Bool) at /home/leopold/.julia/packages/IncrementalInference/ShzPg/src/CliqStateMachineUtils.jl:379
 [3] __doCliqUpSolveInitialized!(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}) at /home/leopold/.julia/packages/IncrementalInference/ShzPg/src/CliqStateMachineUtils.jl:30
 [4] solveUp_StateMachine(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}) at /home/leopold/.julia/packages/IncrementalInference/ShzPg/src/CliqueStateMachine.jl:333
 [5] (::StateMachine{CliqStateMachineContainer})(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}, ::Nothing; pollinterval::Float64, breakafter::Function, verbose::Bool, verbosefid::Base.TTY, verboseXtra::IncrementalInference.CliqStatus, iterlimit::Int64, injectDelayBefore::Nothing, recordhistory::Bool, housekeeping_cb::IncrementalInference.var"#377#379"{IncrementalInference.TreeClique}) at /home/leopold/.julia/packages/FunctionalStateMachine/2JZFG/src/StateMachine.jl:82
 [6] initStartCliqStateMachine!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree, ::IncrementalInference.TreeClique, ::Nothing; oldcliqdata::BayesTreeNodeData, verbose::Bool, verbosefid::Base.TTY, drawtree::Bool, show::Bool, incremental::Bool, limititers::Int64, upsolve::Bool, downsolve::Bool, recordhistory::Bool, delay::Bool, logger::Base.CoreLogging.SimpleLogger, solve_progressbar::ProgressMeter.ProgressUnknown, algorithm::Symbol) at /home/leopold/.julia/packages/IncrementalInference/ShzPg/src/CliqueStateMachine.jl:63
 [7] tryCliqStateMachineSolve!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree, ::Int64, ::Nothing; oldtree::MetaBayesTree, verbose::Bool, verbosefid::Base.TTY, drawtree::Bool, limititers::Int64, downsolve::Bool, incremental::Bool, delaycliqs::Array{Symbol,1}, recordcliqs::Array{Symbol,1}, solve_progressbar::ProgressMeter.ProgressUnknown, algorithm::Symbol) at /home/leopold/.julia/packages/IncrementalInference/ShzPg/src/SolverAPI.jl:110
 [8] (::IncrementalInference.var"#432#435"{MetaBayesTree,Bool,Bool,Base.TTY,Int64,Bool,Bool,Array{Symbol,1},Array{Symbol,1},Symbol,LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree,Nothing,ProgressMeter.ProgressUnknown,Int64})() at ./threadingconstructs.jl:169┌ Warning: printCliqHistorySummary -- No CSM history found.
└ @ IncrementalInference ~/.julia/packages/IncrementalInference/ShzPg/src/TreeDebugTools.jl:211
┌ Warning: printCliqHistorySummary -- No CSM history found.
└ @ IncrementalInference ~/.julia/packages/IncrementalInference/ShzPg/src/TreeDebugTools.jl:211
ERROR: LoadError: TaskFailedException:
RemoteException(2, CapturedException(KeyError(TimeZones [f269a46b-ccf7-5d73-abea-4c690281aa53]), Any[(getindex at dict.jl:467 [inlined], 1), (root_module at loading.jl:968 [inlined], 1), (deserialize_module at Serialization.jl:953, 1), (handle_deserialize at Serialization.jl:855, 1), (deserialize at Serialization.jl:773, 1), (deserialize_datatype at Serialization.jl:1251, 1), (handle_deserialize at Serialization.jl:826, 1), (deserialize at Serialization.jl:773, 1), ( [... cut ...] riableConstraints::Nothing, smtasks::Array{Task,1}, dotreedraw::Array{Int64,1}, runtaskmonitor::Bool, algorithm::Symbol, multithread::Bool) at /home/leopold/.julia/packages/IncrementalInference/ShzPg/src/SolverAPI.jl:371
 [5] top-level scope at /home/leopold/Nextcloud/Studium/Masterarbeit/svn/julia/mmiSAM/examples/speed/speed.jl:118
 [6] top-level scope at timing.jl:233
in expression starting at /home/leopold/Nextcloud/Studium/Masterarbeit/svn/julia/mmiSAM/examples/speed/speed.jl:71

julia> 
```

Affie, 2021-02-10:
Thanks, it looks like this is the relevant part:
```julia
┌ Error: On worker 2:
│ KeyError: key TimeZones [f269a46b-ccf7-5d73-abea-4c690281aa53] not found
│ getindex at ./dict.jl:467 [inlined]
│ root_module at ./loading.jl:968 [inlined]
│ deserialize_module at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Serialization/src/Serialization.jl:953
│ handle_deserialize at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Serialization/src/Serialization.jl:855
│ deserialize at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Serialization/src/Serialization.jl:773
│ deserialize_datatype at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Serialization/src/Serialization.jl:1251
│ handle_deserialize at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Serialization/src/Serialization.jl:826
│ deserialize at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Serialization/src/Serialization.jl:773
│ handle_deserialize at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Serialization/src/Serialization.jl:833
│ deserialize at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Serialization/src/Serialization.jl:1350
│ handle_deserialize at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Serialization/src/Serialization.jl:837
│ deserialize at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Serialization/src/Serialization.jl:773 [inlined]
│ deserialize_dict at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Serialization/src/Serialization.jl:1366
│ deserialize at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Serialization/src/Serialization.jl:1373
│ handle_deserialize at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Serialization/src/Serialization.jl:842
│ deserialize at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Serialization/src/Serialization.jl:1350
│ handle_deserialize at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Serialization/src/Serialization.jl:837
│ deserialize at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Serialization/src/Serialization.jl:1338
│ handle_deserialize at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Serialization/src/Serialization.jl:842
│ deserialize at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Serialization/src/Serialization.jl:773
│ #5 at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Serialization/src/Serialization.jl:932
│ ntuple at ./ntuple.jl:18
│ deserialize_tuple at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Serialization/src/Serialization.jl:932
│ handle_deserialize at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Serialization/src/Serialization.jl:816
│ deserialize at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Serialization/src/Serialization.jl:773 [inlined]
│ deserialize_msg at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Distributed/src/messages.jl:99
│ #invokelatest#1 at ./essentials.jl:710 [inlined]
│ invokelatest at ./essentials.jl:709 [inlined]
│ message_handler_loop at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Distributed/src/process_messages.jl:185
│ process_tcp_streams at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.5/Distributed/src/process_messages.jl:142
│ #99 at ./task.jl:356
```

lemauee, 2021-02-10:
TimeZones.jl or Serialization.jl are appearing a lot in the error message, is there any way that this is related to something on my system besides Julia? Its on Ubuntu 18.04. Or if TimeZones is a problem, anything strange connected to my locale and therfore Timezone?

Affie, 2021-02-10:
I'm also gettting an error in a fresh environment, but not the same:
```julia
caused by: On worker 2:
    TypeError: in new, expected Base.RefValue{Int64}, got a value of type Vector{Float64}
    Stacktrace:
      [1] deserialize
        @ /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Serialization/src/Serialization.jl:1399
      [2] handle_deserialize
        @ /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Serialization/src/Serialization.jl:838
      [3] deserialize
        @ /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Serialization/src/Serialization.jl:774 [inlined]
      [4] deserialize_dict
        @ /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Serialization/src/Serialization.jl:1409
      [5] deserialize
        @ /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Serialization/src/Serialization.jl:1416
      [6] handle_deserialize
        @ /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Serialization/src/Serialization.jl:843
      [7] deserialize
        @ /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Serialization/src/Serialization.jl:1393
      [8] handle_deserialize
        @ /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Serialization/src/Serialization.jl:838
      [9] deserialize
        @ /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Serialization/src/Serialization.jl:1381
     [10] handle_deserialize
        @ /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Serialization/src/Serialization.jl:843
     [11] deserialize
        @ /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Serialization/src/Serialization.jl:774
     [12] #5
        @ /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Serialization/src/Serialization.jl:933
     [13] ntupleany
        @ ./ntuple.jl:43
     [14] deserialize_tuple
        @ /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Serialization/src/Serialization.jl:933
     [15] handle_deserialize
        @ /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Serialization/src/Serialization.jl:817
     [16] deserialize
        @ /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Serialization/src/Serialization.jl:774 [inlined]
     [17] deserialize_msg
        @ /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Distributed/src/messages.jl:87
     [18] #invokelatest#2
        @ ./essentials.jl:707 [inlined]
     [19] invokelatest
        @ ./essentials.jl:706 [inlined]
     [20] message_handler_loop
        @ /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Distributed/src/process_messages.jl:169
     [21] process_tcp_streams
        @ /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.6/Distributed/src/process_messages.jl:126
     [22] #99
        @ ./task.jl:406
```

lemauee, 2021-02-10:
Do you still have Julia 1.5.3 lying around? I see youre on 1.6

Affie, 2021-02-10:
I'm not getting the error on a fresh environment in julia v1.5
```julia
(tmpenv) pkg> st
Status `~/Documents/juliasandbox/tmpenv/Project.toml`
  [91fb55c2] RoME v0.12.2
```

Affie, 2021-02-10:
I can only guess, perhaps try rebuilding TimeZones.jl. Add it to the environment and try `]build TimeZones`

lemauee, 2021-02-10:
Building TimeZones again did the trick :) , but maybe keeping the issue open due to the 1.6 error is still a good thing.

Should the devs of TimeZones or anybody else maybe know about this problem? Having to rebuild stuff manually does not seem like a good workflow ;)

Affie, 2021-02-10:
I'm glad it's working now. Manually building shouldn't be necessary. You can let them know, but it may be hard to reproduce. 
At least anyone with a similar error from RoME can find this thread.

The julia 1.6-rc1 error is not there in RoME#master, so going to close this. It might have been fixed in the latest IIF.

closed 2021-02-10 by Affie

lemauee, 2021-02-10:
Another note on this:

I might have reported this working too early because I had another environment with an old verson of RoME activated when running the "prep-steps". I got the error at some point again. What helped though was activating the right environment on all workers using
```julia
@everywhere using Pkg
@everywhere Pkg.activate(".")  # whatever folder your environment files lie in
```
Maybe this was the source of error all along. Although I don't know If I should have to do this ...

dehann, 2021-02-10:
Activating the right environment on all processes is best practice.  I'm not 100% sure when Pkg.jl uses the same cached library or separates among processes internally.  I have not tested this out too much, but good that we have this data point and time will likely clarify more.  we can link back here as we find more.  Thanks again for posting!

lemauee, 2021-02-17:
As this got me again today, here's a nicer code snippet to activate the same env on all workers:
```julia
@everywhere using Pkg
@everywhere Pkg.activate($(Pkg.project().path))
```

Affie, 2021-02-23:
@lemauee for future reference, it looks like you came across this issue: https://github.com/JuliaLang/julia/issues/28781