JuliaRobotics/RoME.jl #418: Running Rome in multiple julia instances freezes solving

lemauee, 2021-02-25:
Hi,

This is a issue very specific to experiments for my thesis/application:

As I want to perform a lot of individual solves for my thesis, I run multiple julia instances (each using multiple workers and threads) on a pretty beefy machine. This seems to be faster than running them sequentially, because parallelism can't be exploited everywhere. It apparently worked fine until the RoME v0.13 release (I think). Now the solver just "hangs up" in all but one instance (this happens at a different pose for every time I try).

This is the last thing that gets written to stdout (aka my logs ;)):
```julia-repl
Solve Progress: approx max 1752, at iter 711 	 Time: 0:01:50[K[ Info: CSM-5 Clique 47 finished

Solve Progress: approx max 1752, at iter 716 	 Time: 0:01:51[K
```

I run the multiple julia instances each in their own `screen` session like
```bash
screen -S <sessionname>
```
And then fire up my my evaluation script that reads my dataset from a matfile and takes some other arguments (command gets auto-generated by my matlab frontend)
```bash
julia -t auto -p 8 --project=Masterarbeit/svn/julia -J ~/.julia/sysimage_RoME.so -- Masterarbeit/svn/julia/mmiSAM/evaluation/solve2DIncremental.jl --path tmp --file sigmaTrajTf-0-001-0-000-0-005_sigmaLmBearingAndRanging-0-004-0-001_wrongRatio-1-000_minDist-0-200_maxDist-2-000.mat --trajKeys tf --lmKeys bearingAndRanging --startPoseIdx 1 --endPoseIdx 150 --startPoseVal "[7.15793412168699;3.38794983135837;-1.36840501375598]" --plotSaveFinal 1 --plotSaveIter 1 --nRuns 1 --suffix variables --useMsgLikelihoods 0 --nullhypo 0.000000 --tukey 15.000000 --nKernels 100 --spreadNH 3.000000 --inflation 3.000000
```

I will resort to running things sequentally (as any online application would), but especcialy for tuning out parameters like `spreadNH` or `inflation` running things in parallel was very useful, as I don't have access to an infinite amount of machines ;) As multiprocess/multithread performance increased in the last releases, I suspect that something (maybe not even directly RoME-related, but in Julia general) gets in each others way, some lock does not get released.

Best,
Leo

dehann, 2021-02-25:
Hi @lemauee ,

Oops that is odd.  This might be something upstream.  We'd have to look a bit more at what's happening.  In the mean time, perhaps we can help compute results on local servers.  Do you perhaps want to send tar.gz files of graphs you want solved and I can run on compute power here.  Will return the result in new tar.gz.  Perhaps in the process we start getting the same error here.  I'm quite surprised this is multijulia issue is happening.

Sure you already using as such, but just in case:  You can generate with `fg = initfg(); getSolverParams(fg).graphinit=false` and then the generation of graphs should be pretty quick.  Solving can then use either `graphinit` (might not follow exact sequence here) or `treeinit` (our desired but still experimental) method here.

Lastly, I just merged (what I expect) the fix for your issue #378.  Will tag IIF v0.21.3 in a few hours.

cc, might be worth @jim-hill-r (hi!) to look out for something like this multiple Julia's issue too.

Best,
Dehann

lemauee, 2021-02-25:
Hi @dehann,

I don't know if sending you the tar.gz would help to reproduce the issue as expected, as I'm still doing a solve for every pose (I know that this is slow, but consistency to the online-approach we use in classical parametric fgs was important to us). 

However, I'd happily send you the Julia-Project needed to run my stuff, the matfile and the commands to invoke the solve. Otherwise I could also set up the stuff through ssh on your server and then see if i can reproduce it there.

Best,
Leo

lemauee, 2021-02-25:
The same hangup just happened when running only one julia instance, so this might not even be multiinstance-related (but the test yesterday suggested so...). Also, it occured on one of my private and not the universitys pcs. I recently did a fresh ubuntu install there, so nothing strange in the environment should hinder things there. Maybe its also worth to give it a try without a precompiled sysimage as I once had problems with RoMEPlotting as seen in JuliaRobotics/RoMEPlotting.jl#140 as you know.

Best, Leo

lemauee, 2021-02-25:
Whats also notable is that there is absolutely nothing executed (no computational load) present when the hangup occurs (no CPU load in `top` or something similar can be seen).

Affie, 2021-02-25:
Just a sanity check question, did you update the precompiled sysimage with the latest tags?

Other than that, do you know what clique is stalling? 
One way to find out is to draw the tree (drawtree=true, showtree=true) and see at what clique it is waiting on. 
Then you can try turning on debug messages for that clique CSM, say for 2:
`ENV["JULIA_DEBUG"] = :csm_2`
And post results here please.

lemauee, 2021-02-25:
I just compiled the sysimage this morning using RoME v0.13.0 and IIF v0.21.2 . 
The problem is that the clique it stopped on was another one everytime I think, but still need to confirm that. Some examples even ran through that just differed by the amount of noise or even just the setting of the spreadNH parameter and still "made it differently far". I just started a run without the compiled sysimage to roule out that source of failure.
To draw the tree I'll need a machine with x-server to start some pdf viewer, right?

Affie, 2021-02-25:
> To draw the tree I'll need a machine with x-server to start some pdf viewer, right?

You should be able to set `drawtree=true` and then open the tree over SHH, I haven't done it before, so just guessing.

Affie, 2021-02-25:
You could also try setting a timeout  and then seeing if there is some more information on where it timed out, eg:
`solveTree!(fg, timeout=70)`

lemauee, 2021-02-25:
timeout is in seconds I suppose? Then I should give it half an hour (1800s) or something for my example to be safe I'd say ;)
I will start an example with and without sysimage inclding drawtree, showtree and timeout now and report back.

Affie, 2021-02-25:
Seconds, yes. It is per step for every clique individually, so should be fine lower.

lemauee, 2021-02-25:
So lets make it 5 minutes, 300s to be save not to kill something prematurely

Affie, 2021-02-25:
There are a few tools that we use to debug the CSM, you can also try this one:
https://github.com/JuliaRobotics/IncrementalInference.jl/issues/443#issuecomment-699843926

lemauee, 2021-02-25:
Just got an error when trying to run with drawtree, showtree and timeout:
```julia
ERROR: LoadError: MethodError: no method matching solveTree!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree; multithread=true, drawtree=true, showtree=true, timeout=300)
Closest candidates are:
  solveTree!(::AbstractDFG, ::AbstractBayesTree; timeout, storeOld, verbose, verbosefid, delaycliqs, recordcliqs, limititercliqs, injectDelayBefore, skipcliqids, eliminationOrder, variableOrder, eliminationConstraints, variableConstraints, smtasks, dotreedraw, runtaskmonitor, algorithm, multithread) at /home/lemau/.julia/packages/IncrementalInference/nmcd8/src/SolverAPI.jl:258 got unsupported keyword arguments "drawtree", "showtree"
  solveTree!(::AbstractDFG) at /home/lemau/.julia/packages/IncrementalInference/nmcd8/src/SolverAPI.jl:258 got unsupported keyword arguments "multithread", "drawtree", "showtree", "timeout"
Stacktrace:
 [1] kwerr(::NamedTuple{(:multithread, :drawtree, :showtree, :timeout),Tuple{Bool,Bool,Bool,Int64}}, ::Function, ::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree) at ./error.jl:157
```

Affie, 2021-02-25:
Sorry should have been clearer:
```julia
getSolverParams(fg).drawtree = true
getSolverParams(fg).showtree = true
```

lemauee, 2021-02-25:
> There are a few tools that we use to debug the CSM, you can also try this one:
> [JuliaRobotics/IncrementalInference.jl#443 (comment)](https://github.com/JuliaRobotics/IncrementalInference.jl/issues/443#issuecomment-699843926)

This will be difficult if julia "hangs up" right? I dont now how to do the fetching step after the repl is frozen (i would even need to find a way to run my script in the repl with command line arguments first).

Affie, 2021-02-25:
If the timeout is triggered it should stop everything and continue with the next lines.

lemauee, 2021-02-25:
How hard are the performance hits of enabling this? Is there a good way to write this debugging info to file to include it into my normal workflow? Or how to trigger this best on the timeout kicking in?

Affie, 2021-02-25:
I don't think you want it in normal operation. You can remove the `recordcliqs=ls(fg)` or just record a few if you know where the problem is.

Affie, 2021-02-25:
You can add this without a performance hit:
https://github.com/JuliaRobotics/IncrementalInference.jl/issues/443#issuecomment-785895964

lemauee, 2021-02-25:
Opening the `bt.pdf` (using evince) to see the solve progress currently fails on my machine:

Der Dateityp Graphviz-DOT-Graph (text/vnd.graphviz) wird nicht unterstützt

Seems like its no pdf but a dot file.

lemauee, 2021-02-25:
The dot file using the dot viewer seems to be working though :)

lemauee, 2021-02-25:
Also updates fine through sshfs

Affie, 2021-02-25:
You need graphviz and xdot, https://juliarobotics.org/Caesar.jl/latest/installation_environment/#Local-Dependencies

lemauee, 2021-02-25:
Its clearly reproducible when running only one julia instance, I suspect using more workers makes this appear more often.
Tree:
[image]

cliq4_stacktrace:
```julia-repl
InterruptException:
Stacktrace:
 [1] try_yieldto(::typeof(Base.ensure_rescheduled)) at ./task.jl:656
 [2] wait at ./task.jl:713 [inlined]
 [3] wait(::Base.GenericCondition{Base.Threads.SpinLock}) at ./condition.jl:106
 [4] _wait(::Task) at ./task.jl:238
 [5] sync_end(::Channel{Any}) at ./task.jl:294
 [6] macro expansion at ./task.jl:333 [inlined]
 [7] solveCliqDownFrontalProducts!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::IncrementalInference.TreeClique, ::SolverParams, ::Base.CoreLogging.SimpleLogger; MCIters::Int64) at /home/lemau/.julia/packages/IncrementalInference/nmcd8/src/CliqStateMachineUtils.jl:491
 [8] solveCliqDownFrontalProducts! at /home/lemau/.julia/packages/IncrementalInference/nmcd8/src/CliqStateMachineUtils.jl:469 [inlined]
 [9] solveDown_StateMachine(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}) at /home/lemau/.julia/packages/IncrementalInference/nmcd8/src/CliqueStateMachine.jl:654
 [10] (::StateMachine{CliqStateMachineContainer})(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}, ::Int64; pollinterval::Float64, breakafter::Function, verbose::Bool, verbosefid::Base.TTY, verboseXtra::IncrementalInference.CliqStatus, iterlimit::Int64, injectDelayBefore::Nothing, recordhistory::Bool, housekeeping_cb::IncrementalInference.var"#382#384"{IncrementalInference.TreeClique}) at /home/lemau/.julia/packages/FunctionalStateMachine/2JZFG/src/StateMachine.jl:94
 [11] initStartCliqStateMachine!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree, ::IncrementalInference.TreeClique, ::Int64; oldcliqdata::BayesTreeNodeData, verbose::Bool, verbosefid::Base.TTY, drawtree::Bool, show::Bool, incremental::Bool, limititers::Int64, upsolve::Bool, downsolve::Bool, recordhistory::Bool, delay::Bool, logger::Base.CoreLogging.SimpleLogger, solve_progressbar::ProgressMeter.ProgressUnknown, algorithm::Symbol) at /home/lemau/.julia/packages/IncrementalInference/nmcd8/src/CliqueStateMachine.jl:63
 [12] tryCliqStateMachineSolve!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree, ::Int64, ::Int64; oldtree::MetaBayesTree, verbose::Bool, verbosefid::Base.TTY, drawtree::Bool, limititers::Int64, downsolve::Bool, incremental::Bool, delaycliqs::Array{Symbol,1}, recordcliqs::Array{Symbol,1}, solve_progressbar::ProgressMeter.ProgressUnknown, algorithm::Symbol) at /home/lemau/.julia/packages/IncrementalInference/nmcd8/src/SolverAPI.jl:110
 [13] (::IncrementalInference.var"#438#441"{MetaBayesTree,Bool,Bool,Base.TTY,Int64,Bool,Bool,Array{Symbol,1},Array{Symbol,1},Symbol,LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree,Int64,ProgressMeter.ProgressUnknown,Int64})() at ./threadingconstructs.jl:169
```
cliq4_csm.txt is empty

cliq14_stacktrace:
```julia-repl
InterruptException:
Stacktrace:
 [1] try_yieldto(::typeof(Base.ensure_rescheduled)) at ./task.jl:656
 [2] wait at ./task.jl:713 [inlined]
 [3] wait(::Base.GenericCondition{ReentrantLock}) at ./condition.jl:106
 [4] take_unbuffered(::Channel{LikelihoodMessage}) at ./channels.jl:405
 [5] take!(::Channel{LikelihoodMessage}) at ./channels.jl:381
 [6] takeBeliefMessageDown!(::MetaBayesTree, ::LightGraphs.SimpleGraphs.SimpleEdge{Int64}) at /home/lemau/.julia/packages/IncrementalInference/nmcd8/src/TreeMessageAccessors.jl:115
 [7] waitForDown_StateMachine(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}) at /home/lemau/.julia/packages/IncrementalInference/nmcd8/src/CliqueStateMachine.jl:467
 [8] (::StateMachine{CliqStateMachineContainer})(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}, ::Int64; pollinterval::Float64, breakafter::Function, verbose::Bool, verbosefid::Base.TTY, verboseXtra::IncrementalInference.CliqStatus, iterlimit::Int64, injectDelayBefore::Nothing, recordhistory::Bool, housekeeping_cb::IncrementalInference.var"#382#384"{IncrementalInference.TreeClique}) at /home/lemau/.julia/packages/FunctionalStateMachine/2JZFG/src/StateMachine.jl:94
 [9] initStartCliqStateMachine!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree, ::IncrementalInference.TreeClique, ::Int64; oldcliqdata::BayesTreeNodeData, verbose::Bool, verbosefid::Base.TTY, drawtree::Bool, show::Bool, incremental::Bool, limititers::Int64, upsolve::Bool, downsolve::Bool, recordhistory::Bool, delay::Bool, logger::Base.CoreLogging.SimpleLogger, solve_progressbar::ProgressMeter.ProgressUnknown, algorithm::Symbol) at /home/lemau/.julia/packages/IncrementalInference/nmcd8/src/CliqueStateMachine.jl:63
 [10] tryCliqStateMachineSolve!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree, ::Int64, ::Int64; oldtree::MetaBayesTree, verbose::Bool, verbosefid::Base.TTY, drawtree::Bool, limititers::Int64, downsolve::Bool, incremental::Bool, delaycliqs::Array{Symbol,1}, recordcliqs::Array{Symbol,1}, solve_progressbar::ProgressMeter.ProgressUnknown, algorithm::Symbol) at /home/lemau/.julia/packages/IncrementalInference/nmcd8/src/SolverAPI.jl:110
[... cut ...]
 [6] macro expansion at ./timing.jl:233 [inlined]
 [7] top-level scope at /home/lemau/Masterarbeit/svn/julia/mmiSAM/evaluation/solve2DIncremental.jl:159
 [8] include(::Function, ::Module, ::String) at ./Base.jl:380
 [9] include(::Module, ::String) at ./Base.jl:368
 [10] exec_options(::Base.JLOptions) at ./client.jl:296
 [11] _start() at ./client.jl:506
in expression starting at /home/lemau/Masterarbeit/svn/julia/mmiSAM/evaluation/solve2DIncremental.jl:128
lemau@joule:~$ 

```

I will run it again and see if it fails at a different spot.

Affie, 2021-02-25:
Can you upload the saved factor graph for the above example and I'll see if I can reproduce it.

Affie, 2021-02-25:
It is clique 4 that is timing out on the down solve. 
We haven't seen something like this before.

lemauee, 2021-02-25:
Unfortunately as I ran this as a script and not from the REPL (as I would have to look into how to supply all the command line arguments to my script in the REPL) I only have the "pose-iteration" before the failure. I can inform myself (or do you know an easy way how to do this) and do this in a later run and save it out, but I'd wait for the repetition I run now (should be finished in 1-2 hours if it breaks at the same spot) to see if it breaks at the same spot.

Affie, 2021-02-25:
Another thing you could try is to turn on debug, I think it doesn't work well with `multithread=true` yet, so perhaps switch it off.
```julia
getSolverParams(fg).dbg=true
````
It's also possible that `multithread=true` is causing your problem.

Affie, 2021-02-25:
> Unfortunately as I ran this as a script and not from the REPL (as I would have to look into how to supply all the command line arguments to my script in the REPL) I only have the "pose-iteration" [...]

Perhaps save the fg before every solve in the script (if I'm understanding you correctly?)

lemauee, 2021-02-25:
The second run already got past pose 50 it reached in the first one, so my suspicion that this is not related to a particular graph seems true.

Not using my sysimage also did not help, timeout/hangup also occur there.

lemauee, 2021-02-25:
> Perhaps save the fg before every solve in the script (if I'm understanding you correctly?)

Why didnt this obvious solution occur to me ;) Saving the fg before every solve would be possible, but since the graph itself doesn't seem to be causing the problem, I'll try other things instead.

lemauee, 2021-02-25:
> Another thing you could try is to turn on debug, I think it doesn't work well with `multithread=true` yet, so perhaps switch it off.
> 
> ```julia
> getSolverParams(fg).dbg=true
> ```
> 
> It's also possible that `multithread=true` is causing your problem.

Switching `multitheread` off and maybe even running it on one worker are my next best ideas to track this down.

Affie, 2021-02-25:
In that case perhaps `getSolverParams(fg).dbg=true` will work, the subgraph of every clique is saved before the solve. If it fails you can re-run the subgraph and see if it's repeatable. 
It will also save the last state of the CSM. I'll have to find the code to extract it (can't remember).

Affie, 2021-02-25:
Also perhaps you can rebuild the 50 pose fg that failed and share it. We can queue a few solves and see if all of them pass.

lemauee, 2021-02-25:
As the second run already got past pose 50, I also have the graph for this ;)

[00050_fg.tar.gz](https://github.com/JuliaRobotics/RoME.jl/files/6044533/00050_fg.tar.gz)

lemauee, 2021-02-25:
> In that case perhaps `getSolverParams(fg).dbg=true` will work, the subgraph of every clique is saved before the solve. If it fails you can re-run the subgraph and see if it's repeatable.
> It will also save the last state of the CSM. I'll have to find the code to extract it (can't remember).
> 
> EDIT:
> With debug on, for failed task `i` the last state can be extracted with `smtasks[i].storage[:csmc]`
> I think you will have to do it locally or start julia in the remote to have access to the REPL for debugging and saving.

Is it possible to save this to file or should I start to search how I can pass command line Arguments to a script in the REPL?

Affie, 2021-02-25:
The subgraph will be stored in a file per clique. (default somewhere in /tmp/caesar)

The task storage files will have to be debugged interactively. You can try and save it with https://docs.julialang.org/en/v1/stdlib/Serialization/ but i don't know how portable it will be.

lemauee, 2021-02-25:
I'll read into passing Command line arguments in the REPL then. 

The 'multithread off' and 'completely on one core' examples are running now, I'll see how that goes. But we really have to look out for 'lucky shots': one solve just failed at pose 139/150 (running on 8 not 16 workers because of different hardware underneath), this easily could have finished without something going wrong out of pure luck ;) Maybe using a lot of workers makes this fail earlier?

dehann, 2021-02-26:
likely a result of a rare bug and therefore I suggest using instead  `getSolverParams(fg).useMsgLikelihood=false` for the time being.  I don't think this is related to multithreading as speculated earlier, however, testing to confirm is ongoing.  See here for more details:
https://github.com/JuliaRobotics/IncrementalInference.jl/issues/1136#issuecomment-786326876

Affie, 2021-02-26:
@dehann, I'm almost certain it's not related, he is not getting an error. We are not serializing solver parameters and we need to set them manually to the same values Leo is using. I only know he using.useMsgLiklihoods=false`, the rest might also have an influence here. From his script:
> `--useMsgLikelihoods 0 --spreadNH 3.000000 --inflation 3.000000`

@lemauee could you please write down the solver parameters that were set with the factor graphs saved.

dehann, 2021-02-26:
oh okay, yes please send the solver parameters used.

----
Related 

xref https://github.com/JuliaRobotics/IncrementalInference.jl/issues/1183#issuecomment-785305730

Affie, 2021-02-26:
It does not look like we are able to reproduce this issue on our side. I ran it a few times overnight and from slack I see Dehann ran it a lot too. 
I was on julia 1.6-rc1 IIF 0.21.3 and RoME v0.13
The best I can think of is to help figure out how you can debug it yourself. 

On the command line arguments question, I don't know if [ARGS](https://docs.julialang.org/en/v1/base/constants/#Base.ARGS) is what you mean, it looks like you are already using it. 
Perhaps you can use a bash script?

lemauee, 2021-02-26:
> oh okay, yes please send the solver parameters used.
> 
> Related
> 
> xref [JuliaRobotics/IncrementalInference.jl#1183 (comment)](https://github.com/JuliaRobotics/IncrementalInference.jl/issues/1183#issuecomment-785305730)

The solver params used are actually a little different for every example (as I was about to tune out the best spreadNH for my usecase in some of the calculations that failed): I always have `useMsgLikelihoods=false` though and `inflation=3.0` (which should be the standard value, maybe I could even add a -1 in my commandline interface to use the standard parameter for this in the script itself). `spreadNH` was set to 3.0, 4.0, 5.0 and 6.0 depending on the example.

lemauee, 2021-02-26:
I have not updated from IIF 0.21.2 to .3 yet, but could do this of course.

For the commandline stuff, its possible to set ARGS in the repl. I made myself a printargs script that only does
```julia
print(ARGS);
```
and should be able to set args in the repl using the output of this script called with my cli arguments. So I should be able to use the clique debugging stuff in my script if the last changes I did do not solve my problems.

lemauee, 2021-02-26:
Even the example using only one thread on one worker with solveTree!(...,multithread=false), addFactor!(...,threadModel=SingleThreaded)  and no `@async` call failed. I will run it in the REPL now to get the debug stuff.

lemauee, 2021-02-26:
https://github.com/lemauee/thesis_calculations should reproduce the error

lemauee, 2021-02-27:
Hi, I now have a failed example in the repl with 

> You can add this without a performance hit:
> [JuliaRobotics/IncrementalInference.jl#443 (comment)](https://github.com/JuliaRobotics/IncrementalInference.jl/issues/443#issuecomment-785895964)

enabled:

[image]

 What steps could I do now to debug it / send you data to look at? Serializing the smtasks and even individual smtasks[i].storage[:csmc] fails:

[image]

 I could only send you the smtasks[i].storage[:statemachine] files (which at least cover the state the csm stopped in, right?) 

Another problem I have is that my REPL still gets spammed by a Fontconfig error (this is why I sent you screenshots instead of code as otherwise (I'm not fast enought to copy what I see in the REPL before this gets overwritten). It seems like there is still something running, but I dont have much CPU load from the julia process. The Fontconfig error has long been present, but I didn't bother to resolve it since everything seemed to work. But It might even be that this is related to the original freezing problem now.
[image]

Best,
Leo

Affie, 2021-02-27:
Hi Leo. I’m not in front of a computer, but can try and help. 
The repl messages can come from drawtree, the task is probably not stopped cleanly with the async option on (its best not to use drawtree with async for now as it still needs better api to control it) you can lookin the solverapi.jl source code and try to find the draw task, but i cant remember now. 

For debugging, there is a subfg object in :csmc. You can save that. The `:statemachine` Has the last function called. The way to debug would be to rerun the function on csmc. Perhaps turn on debugging for that clique with ENV as mentioned above.

lemauee, 2021-02-27:
Hi @Affie,
according to :statemachine all cliques seem to be stuck on the same function:
```julia
julia> using RoME
[ Info: Including InteractiveUtils related functions in IncrementalInference.

julia> using Serialization

julia> deserialize("statemachine58.out")
StateMachine{CliqStateMachineContainer}(IncrementalInference.waitForDown_StateMachine, 6, Tuple{Dates.DateTime,Int64,Function,CliqStateMachineContainer}[], "cliq58")

julia> deserialize("statemachine55.out")
StateMachine{CliqStateMachineContainer}(IncrementalInference.waitForDown_StateMachine, 6, Tuple{Dates.DateTime,Int64,Function,CliqStateMachineContainer}[], "cliq55")

julia> deserialize("statemachine54.out")
StateMachine{CliqStateMachineContainer}(IncrementalInference.waitForDown_StateMachine, 6, Tuple{Dates.DateTime,Int64,Function,CliqStateMachineContainer}[], "cliq54")

julia> deserialize("statemachine53.out")
StateMachine{CliqStateMachineContainer}(IncrementalInference.waitForDown_StateMachine, 6, Tuple{Dates.DateTime,Int64,Function,CliqStateMachineContainer}[], "cliq53")

julia> deserialize("statemachine52.out")
StateMachine{CliqStateMachineContainer}(IncrementalInference.waitForDown_StateMachine, 6, Tuple{Dates.DateTime,Int64,Function,CliqStateMachineContainer}[], "cliq52")

julia> deserialize("statemachine44.out")
StateMachine{CliqStateMachineContainer}(IncrementalInference.waitForDown_StateMachine, 6, Tuple{Dates.DateTime,Int64,Function,CliqStateMachineContainer}[], "cliq44")
```
The tree implies the same:
[image]

lemauee, 2021-02-27:
I think i didn't even turn on `getSolverParams(fg).async = true`, is it on by default?

lemauee, 2021-02-27:
In the (still messed up) REPL I can't get a the subfg object with `csmc58sfg = smtasks[58].storage[:csmc].subfg`, does it have a different fieldname or is the storage[:csmc] another dict? Could we rerun the function on this or do we need the whole csmc object?

I also thougth about quitting the REPL session again to run it again with async, drawtree and showtree off to be able to work in that repl properly ;)

Affie, 2021-02-27:
Sorry, I’m going from memory. I think it’s `cliqSubFg`. Can you try a deepcopy and then serialize the whole csmc object. We do need it, but can try and rebuild it if it doesn’t want to work.

lemauee, 2021-02-27:
Just as a note for writing the somehow protechted ARGS, this is the stuff I do now:

```julia-repl
julia> newARGS=["--path", "tmp2", "--file", "sigmaTrajTf-0-000-0-000-0-003_sigmaLmBearingAndRanging-0-002-0-001.mat", "--trajKeys", "tf", "--lmKeys", "bearingAndRanging", "--startPoseIdx", "1", "--endPoseIdx", "150", "--startPoseVal", "[7.15793412168699;3.38794983135837;-1.36840501375598]", "--plotSaveFinal", "1", "--plotSaveIter", "1", "--nRuns", "1", "--suffix", "variables", "--useMsgLikelihoods", "0", "--nullhypo", "0.000000", "--tukey", "15.000000", "--nKernels", "100", "--spreadNH", "3.000000", "--inflation", "3.000000"]
julia> empty!(ARGS)
julia> for arg in newARGS; push!(ARGS,arg); end
```

A new session with drawtree/showtree/async off is running now, but I still have the old one online.

EDIT: when running the new session on the same machine with the old one still on another screen, I get this error:

```julia
ERROR: LoadError: TaskFailedException:
IOError: mkdir: permission denied (EACCES)
Stacktrace:
 [1] uv_error at ./libuv.jl:97 [inlined]
 [2] mkdir(::String; mode::UInt16) at ./file.jl:177
 [3] mkpath(::String; mode::UInt16) at ./file.jl:227
 [4] mkpath(::String; mode::UInt16) at ./file.jl:225 (repeats 4 times)
 [5] mkpath at ./file.jl:222 [inlined]
 [6] drawGraph(::LightDFG{SolverParams,DFGVariable,DFGFactor}; viewerapp::String, filepath::String, engine::String, show::Bool) at /home/leopold/.julia/packages/IncrementalInference/nmcd8/src/AdditionalUtils.jl:56
 [7] _dbgCSMSaveSubFG(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}, ::String) at /home/leopold/.julia/packages/IncrementalInference/nmcd8/src/CliqStateMachineUtils.jl:75
 [8] buildCliqSubgraph_StateMachine(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}) at /home/leopold/.julia/packages/IncrementalInference/nmcd8/src/CliqueStateMachine.jl:122
 [9] (::StateMachine{CliqStateMachineContainer})(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}, ::Int64; pollinterval::Float64, breakafter::Function, verbose::Bool, verbosefid::Base.TTY, verboseXtra::IncrementalInference.CliqStatus, iterlimit::Int64, injectDelayBefore::Nothing, recordhistory::Bool, housekeeping_cb::IncrementalInference.var"#382#384"{IncrementalInference.TreeClique}) at /home/leopold/.julia/packages/FunctionalStateMachine/2JZFG/src/StateMachine.jl:94
 [10] initStartCliqStateMachine!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree, ::IncrementalInference.TreeClique, ::Int64; oldcliqdata::BayesTreeNodeData, verbose::Bool, verbosefid::Base.TTY, drawtree::Bool, show::Bool, incremental::Bool, limititers::Int64, upsolve::Bool, downsolve::Bool, recordhistory::Bool, delay::Bool, logger::Base.CoreLogging.SimpleLogger, solve_progressbar::ProgressMeter.ProgressUnknown, algorithm::Symbol) at /home/leopold/.julia/packages/IncrementalInference/nmcd8/src/CliqueStateMachine.jl:63
 [11] tryCliqStateMachineSolve!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree, ::Int64, ::Int64; oldtree::MetaBayesTree, verbose::Bool, verbosefid::Base.TTY, drawtree::Bool, limititers::Int64, downsolve::Bool, incremental::Bool, delaycliqs::Array{Symbol,1}, recordcliqs::Array{Symbol,1}, solve_progressbar::ProgressMeter.ProgressUnknown, algorithm::Symbol) at /home/leopold/.julia/packages/IncrementalInference/nmcd8/src/SolverAPI.jl:110
 [12] (::IncrementalInference.var"#439#442"{MetaBayesTree,Bool,Bool,Base.TTY,Bool,Bool,Array{Symbol,1},Array{Symbol,1},Symbol,LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree,Int64,ProgressMeter.ProgressUnknown,Int64,Int64})() at ./task.jl:356
Stacktrace:
[ Info: monitorCSMs: all tasks done
 [1] sync_end(::Channel{Any}) at ./task.jl:314
 [2] macro expansion at ./task.jl:333 [inlined]
 [3] taskSolveTree!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree, ::Int64; oldtree::MetaBayesTree, drawtree::Bool, verbose::Bool, verbosefid::Base.TTY, limititers::Int64, limititercliqs::Array{Pair{Symbol,Int64},1}, downsolve::Bool, incremental::Bool, multithread::Bool, skipcliqids::Array{Symbol,1}, recordcliqs::Array{Symbol,1}, delaycliqs::Array{Symbol,1}, smtasks::Array{Task,1}, algorithm::Symbol) at /home/leopold/.julia/packages/IncrementalInference/nmcd8/src/SolverAPI.jl:49
 [4] solveTree!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree; timeout::Int64, storeOld::Bool, verbose::Bool, verbosefid::Base.TTY, delaycliqs::Array{Symbol,1}, recordcliqs::Array{Symbol,1}, limititercliqs::Array{Pair{Symbol,Int64},1}, injectDelayBefore::Nothing, skipcliqids::Array{Symbol,1}, eliminationOrder::Nothing, variableOrder::Nothing, eliminationConstraints::Array{Symbol,1}, variableConstraints::Nothing, smtasks::Array{Task,1}, dotreedraw::Array{Int64,1}, runtaskmonitor::Bool, algorithm::Symbol, multithread::Bool) at /home/leopold/.julia/packages/IncrementalInference/nmcd8/src/SolverAPI.jl:371
 [5] macro expansion at /home/leopold/Masterarbeit/svn/julia/mmiSAM/evaluation/solve2DIncremental.jl:156 [inlined]
 [6] macro expansion at ./timing.jl:233 [inlined]
 [7] top-level scope at /home/leopold/Masterarbeit/svn/julia/mmiSAM/evaluation/solve2DIncremental.jl:149
 [8] include(::String) at ./client.jl:457
 [9] top-level scope at REPL[4]:1
in expression starting at /home/leopold/Masterarbeit/svn/julia/mmiSAM/evaluation/solve2DIncremental.jl:131
```

Should be related to the `getSolverParams(fg).dbg=true` still set.

lemauee, 2021-02-27:
> Sorry, I’m going from memory. I think it’s `cliqSubFg`. Can you try a deepcopy and then serialize the whole csmc object. We do need it, but can try and rebuild it if it doesn’t want to work.

serializing .cliqSubFg works :)

lemauee, 2021-02-27:
Doing the deepcopy and then serializing the csmc container did not work:
csmc58dc = deepcopy(csmc58)
[image]

lemauee, 2021-02-27:
I will have to quit the session and run again or run on another machine then if theres no way to stop the draw task.

Affie, 2021-02-27:
With the task that doesn’t want to serialize, you can try throwing an interrupt exception to it.

lemauee, 2021-02-27:
> Just as a note for writing the somehow protechted ARGS, this is the stuff I do now:
> 
> ```
> julia> newARGS=["--path", "tmp2", "--file", "sigmaTrajTf-0-000-0-000-0-003_sigmaLmBearingAndRanging-0-002-0-001.mat", "--trajKeys", "tf", "--lmKeys", "bearingAndRanging", "--startPoseIdx", "1", "--endPoseIdx", "150", "--startPoseVal", "[7.15793412168699;3.38794983135837;-1.36840501375598]", "--plotSaveFinal", "1", "--plotSaveIter", "1", "--nRuns", "1", "--suffix", "variables", "--useMsgLikelihoods", "0", "--nullhypo", "0.000000", "--tukey", "15.000000", "--nKernels", "100", "--spreadNH", "3.000000", "--inflation", "3.000000"]
> julia> empty!(ARGS)
> julia> for arg in newARGS; push!(ARGS,arg); end
> ```
> 
> A new session with drawtree/showtree/async off is running now, but I still have the old one online.
> 
> EDIT: when running the new session on the same machine with the old one still on another screen, I get this error:
> 
> ```julia
> ERROR: LoadError: TaskFailedException:
> IOError: mkdir: permission denied (EACCES)
> Stacktrace:
>  [1] uv_error at ./libuv.jl:97 [inlined]
>  [2] mkdir(::String; mode::UInt16) at ./file.jl:177
>  [3] mkpath(::String; mode::UInt16) at ./file.jl:227
>  [4] mkpath(::String; mode::UInt16) at ./file.jl:225 (repeats 4 times)
>  [5] mkpath at ./file.jl:222 [inlined]
>  [6] drawGraph(::LightDFG{SolverParams,DFGVariable,DFGFactor}; viewerapp::String, filepath::String, engine::String, show::Bool) at /home/leopold/.julia/packages/IncrementalInference/nmcd8/src/AdditionalUtils.jl:56
>  [7] _dbgCSMSaveSubFG(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}, ::String) at /home/leopold/.julia/packages/IncrementalInference/nmcd8/src/CliqStateMachineUtils.jl:75
>  [8] buildCliqSubgraph_StateMachine(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}) at /home/leopold/.julia/packages/IncrementalInference/nmcd8/src/CliqueStateMachine.jl:122
>  [9] (::StateMachine{CliqStateMachineContainer})(::CliqStateMachineContainer{BayesTreeNodeData,LightDFG{SolverParams,DFGVariable,DFGFactor},LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree}, ::Int64; pollinterval::Float64, breakafter::Function, verbose::Bool, verbosefid::Base.TTY, verboseXtra::IncrementalInference.CliqStatus, iterlimit::Int64, injectDelayBefore::Nothing, recordhistory::Bool, housekeeping_cb::IncrementalInference.var"#382#384"{IncrementalInference.TreeClique}) at /home/leopold/.julia/packages/FunctionalStateMachine/2JZFG/src/StateMachine.jl:94
>  [10] initStartCliqStateMachine!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree, ::IncrementalInference.TreeClique, ::Int64; oldcliqdata::BayesTreeNodeData, verbose::Bool, verbosefid::Base.TTY, drawtree::Bool, show::Bool, incremental::Bool, limititers::Int64, upsolve::Bool, downsolve::Bool, recordhistory::Bool, delay::Bool, logger::Base.CoreLogging.SimpleLogger, solve_progressbar::ProgressMeter.ProgressUnknown, algorithm::Symbol) at /home/leopold/.julia/packages/IncrementalInference/nmcd8/src/CliqueStateMachine.jl:63
>  [11] tryCliqStateMachineSolve!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree, ::Int64, ::Int64; oldtree::MetaBayesTree, verbose::Bool, verbosefid::Base.TTY, drawtree::Bool, limititers::Int64, downsolve::Bool, incremental::Bool, delaycliqs::Array{Symbol,1}, recordcliqs::Array{Symbol,1}, solve_progressbar::ProgressMeter.ProgressUnknown, algorithm::Symbol) at /home/leopold/.julia/packages/IncrementalInference/nmcd8/src/SolverAPI.jl:110
>  [12] (::IncrementalInference.var"#439#442"{MetaBayesTree,Bool,Bool,Base.TTY,Bool,Bool,Array{Symbol,1},Array{Symbol,1},Symbol,LightDFG{SolverParams,DFGVariable,DFGFactor},MetaBayesTree,Int64,ProgressMeter.ProgressUnknown,Int64,Int64})() at ./task.jl:356
> Stacktrace:
> [ Info: monitorCSMs: all tasks done
>  [1] sync_end(::Channel{Any}) at ./task.jl:314
>  [2] macro expansion at ./task.jl:333 [inlined]
>  [3] taskSolveTree!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree, ::Int64; oldtree::MetaBayesTree, drawtree::Bool, verbose::Bool, verbosefid::Base.TTY, limititers::Int64, limititercliqs::Array{Pair{Symbol,Int64},1}, downsolve::Bool, incremental::Bool, multithread::Bool, skipcliqids::Array{Symbol,1}, recordcliqs::Array{Symbol,1}, delaycliqs::Array{Symbol,1}, smtasks::Array{Task,1}, algorithm::Symbol) at /home/leopold/.julia/packages/IncrementalInference/nmcd8/src/SolverAPI.jl:49
>  [4] solveTree!(::LightDFG{SolverParams,DFGVariable,DFGFactor}, ::MetaBayesTree; timeout::Int64, storeOld::Bool, verbose::Bool, verbosefid::Base.TTY, delaycliqs::Array{Symbol,1}, recordcliqs::Array{Symbol,1}, limititercliqs::Array{Pair{Symbol,Int64},1}, injectDelayBefore::Nothing, skipcliqids::Array{Symbol,1}, eliminationOrder::Nothing, variableOrder::Nothing, eliminationConstraints::Array{Symbol,1}, variableConstraints::Nothing, smtasks::Array{Task,1}, dotreedraw::Array{Int64,1}, runtaskmonitor::Bool, algorithm::Symbol, multithread::Bool) at /home/leopold/.julia/packages/IncrementalInference/nmcd8/src/SolverAPI.jl:371
>  [5] macro expansion at /home/leopold/Masterarbeit/svn/julia/mmiSAM/evaluation/solve2DIncremental.jl:156 [inlined]
>  [6] macro expansion at ./timing.jl:233 [inlined]
>  [7] top-level scope at /home/leopold/Masterarbeit/svn/julia/mmiSAM/evaluation/solve2DIncremental.jl:149
>  [8] include(::String) at ./client.jl:457
>  [9] top-level scope at REPL[4]:1
> in expression starting at /home/leopold/Masterarbeit/svn/julia/mmiSAM/evaluation/solve2DIncremental.jl:131
> ```
> 
> Should be related to the `getSolverParams(fg).dbg=true` still set.

Also appears on another machine.

lemauee, 2021-02-27:
> With the task that doesn’t want to serialize, you can try throwing an interrupt exception to it.

If the smtasks are already done or failed, should that be necessary? Or are we talking about a different task here? How do you throw it and at what task exactly?

Affie, 2021-02-27:
I’m just going by the error message of: cannot serialize a running task.  Mabe the clique sub fg is enough to figure it out.

[13 more replies]