JuliaRobotics/RoME.jl #450: Should a Lazy wrapper for Manifolds.jl exist?

Affie, 2021-06-14:
Hi @Affie, i'm having doubts about this whole approach.  I'm now thinking we should not be building types like SE and se at all.

I think we should drastically reduce the amount of duplication with Manifolds.jl and just simply use 
VND.val::Vector{ProductRepr...}

and implement just RoME.oplus(::Pose3Pose3, p1,p2) which follows the same style as Manifolds.jl.  That way we avoid the exp/retract issue we had in 366

I really want to avoid duplicating types, abstracts and API design from Manifolds.  I'm actually thinking we should not implement a Lazy wrapper at all for the next couple of weeks and simply add the oplus functions in RoME until we have a better idea of what does and doesn't work.  All we doing is dispatch against func(::AbstractFactor,...). E.g. 
v = getCoordinates(Pose3, p1)

which is easily overloaded for
v_arr = getPointCoordinates(fg, :x1) # which we already know is a Pose2 or InertialPose3 etc

_Originally posted by @dehann in https://github.com/dehann/TransformUtils.jl/issues/45#issuecomment-860329987_

Affie, 2021-06-14:
The main reason I like a wrapper is to give concrete types as there is no type distinction between representations of the group and the algebra.

> RoME.oplus(::Pose3Pose3, p1,p2) which follows the same style as Manifolds.jl

I wouldn't implement this without concrete types on p1 and p2. 
as you also have
`RoME.oplus(::Pose3Pose3, p,X)` and `RoME.oplus(::Pose3Pose3, X, p)`
And oplus can then also be generalized to any lie group RoME.oplus(M::AbstractGroupManifold, p1,p2)

The second reason, although not really that important, is it makes it slightly easier for me to use. 
Take PosePose as an example:
```julia
function PosePose(ⁱmⱼ::SE{N}, ʷxᵢ::SE{N}, ʷxⱼ::SE{N}) where N
    ʷT̂ⱼ = ʷxᵢ ∘ ⁱmⱼ
    ʲT̂ⱼ = ʷxⱼ \ ʷT̂ⱼ
    X = log(ʲT̂ⱼ)
    return vee(X)
end
```
The equivalent will be something like this to implement Pose2 and Pose3:
```julia
function Pose2Pose2(ⁱmⱼ, ʷxᵢ, ʷxⱼ)
    M = SpecialEuclidean(2)
    ʷT̂ⱼ = compose(M, ʷxᵢ, ⁱmⱼ)
    ʲT̂ⱼ = compose(M, inv(M, ʷxⱼ), ʷT̂ⱼ)
    X = log(M, identity(M,ʲT̂ⱼ), ʲT̂ⱼ)
    return vee(M, identity(M,ʲT̂ⱼ), X)
end

function Pose3Pose3(ⁱmⱼ, ʷxᵢ, ʷxⱼ)
    M = SpecialEuclidean(3)
    ʷT̂ⱼ = compose(M, ʷxᵢ, ⁱmⱼ)
    ʲT̂ⱼ = compose(M, inv(M, ʷxⱼ), ʷT̂ⱼ)
    X = log(M, identity(M,ʲT̂ⱼ), ʲT̂ⱼ)
    return vee(M, identity(M,ʲT̂ⱼ), X)
end
``` 
A third reason is the constructors that are currently in TransformUtils.  eg. `SO3(E::EulerAngles)`

-----------------------------------------

That being said, I'm also not completely convinced either and AMP is likely the place that will determine what is needed.

I would also like to explore the possibility of the variable actually being the type. eg.
```julia
struct Pose{N <: Manifolds.AbstractGroupManifold, T}
  value::T 
end
```

Affie, 2021-06-15:
In a similar way, the data type and manifold can be stored with the InferenceVariable.

```julia
abstract type InferenceVariable{M<:AbstractManifold, T} end

struct Pose{M<:SpecialEuclidean, T} <: InferenceVariable{M, T} end

const Pose2 = Pose{SpecialEuclidean{2}, ProductRepr{Tuple{SVector{2, Float64}, SMatrix{2, 2, Float64, 4}}}}
const Pose3 = Pose{SpecialEuclidean{3}, ProductRepr{Tuple{SVector{3, Float64}, SMatrix{3, 3, Float64, 9}}}}

T = Pose{SpecialEuclidean{2}, SVector{3,Float64}}()

getPointType(::InferenceVariable{M, T}) where {M,T} = T
getManifoldType(::InferenceVariable{M, T}) where {M,T} = M

vnd::Vector{getPointType(T)}()
```

dehann, 2021-06-15:
Hi Johan, thanks for the write-up, I have many thoughts and will respond later today or tomorrow once I get the ideas organized.  I think there is an Option D which I favor, but let me go make sure first.  Thanks again for the write-up!

Affie, 2021-06-16:
Rabbit hole: To capture it somewhere:
As expected but still cool to see, all factors from a point to a point on group manifolds will look exactly the same (also priors)

```julia
#manifolds point to manifold point factor _d uses distance function
function PointPoint_d(m, p, q)
    q̂ = p ∘ m
    return distance(q, q̂)
end
# manifold prior factor _d uses distance function
function Prior_d(meas, p)		
    return distance(meas, p)
end
```

dehann, 2021-06-17:
## Option D

Stay with `DFG.@defVariable` (and few small modifications) so that these functions remain:
```julia
getDimension(::Manifold)
getManifold(::InferenceVariable)
```

This resolves through dispatch how "Manifold type" information is retrieved -- note, not the "point/element type" information.  Must resolve that separately.  For serialization, we expressly decided against any value information inside `VariableType`, which immediately prevents option A and B (JuliaRobotics/IncrementalInference.jl#783, and JuliaRobotics/DistributedFactorGraphs.jl#603).

Serialization of variable types for Option C is also hard, but that is easier to just use the `getManifolds` dispatch as well.

@Affie and I spoke on the phone, and a workable solution for resolving "point type" information is to include this (from the user) via:
```juila
@defVariable Pose3 Manifolds.SpecialEuclidean(3) typeof(Manifolds.ProductRepr(MVector3.., MMatrix33...)
```
i.e. both "manifold type" and "point type" information is provided by the user.  

Most important though, is to **NOT** require other new abstract definitions or wrapper types for existing Manifolds types.  Once we have nonparametric and parametric computations fully consolidated on Manifolds.jl operations (i.e. deprecate TransformUtils.jl), then we can as a later step return with wrappers to help (e.g. `LazyLieSE2Point{Float64}`):
```julia
@defVariable Pose3_EZ Manifolds.SpecialEuclidean(3) LazyLieSE2Point{Float64}

# or even do a new macro
@easyDefVariable LazyLieSE2Point{Float64}
```
But this is only a later step **AFTER** IIF has been transitioned to Manifolds without adding new types and abstractions.

I'm trying to keep both maximum flexibility and lesson learned inside Manifolds.jl available here (preserving utility and unity in Julia ecosystem), while also minimizing the changes in IIF/RoME.  I understand Options A,B,C above as "already assume wrappers are the way to go".   I originally thought we should go that way, but now I'm more convinced that this Option D is the necessary first step -- otherwise we are going to take on way too much work at once.

Affie, 2021-06-18:
In summary:

We are going with option D, which is option A but with a Lazy wrapper only implemented at a later stage if needed. 

In options B and C, the lazy wrapper was implemented in the `<:InferenceVariable` types. With serialization to remain as is, it sounds like a bad idea. 

I'm pretty sure a lazy wrapper (or a place to extend Manifolds.jl) will come in handy later.