JuliaRobotics/RoME.jl #465: Design for Manifolds.jl factors upgrade

Affie, 2021-07-29:
Currenlty we have:
`getSample`
 - Priors return points on manifold
 - Factors return tangent vector (but can be anything as long as it matches the factor)

`AbstractManifoldMinimize` should return a tangent vector
`AbstractRelativeRoots` and `AbstractRelitiveMinimize` returns coordinates and should be updated to take a point on manifold as input.

Affie, 2021-08-02:
The current implementation has conflicts with the way `getSample` works for `MKD` vs `<:Distributions` 
Rand on `MKD` returns a point on the manifold where `getSample` expects a coordinate.

dehann, 2021-08-02:
Yes, `rand(::MKD)::Vector{P}` is unavoidable I think.  Distributions.jl assumes that all samples will always be a matrix with coordinates as columns.  We are moving past that.  My suggestions is we just make IIF always expect `Vector{P}`.  This is how I fixed it:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/db7ff84225cc848c325e57b5fb9d0d85cb6c79b8/src/SolverUtilities.jl#L12-L22

I have the IIF objective as:
- all factors residual functions must take in manifold points (without forcing any type constraints on arguments),
- `<:AbstractRelativeMinimize/Roots` residual calc must return residual in coordinates,
- `<:AbstractManifoldMinimize` residual calc must return residual as a tangent (vector), 
- `getSample` for `<:AbstractRelative` can return manifolds-either-or -- i.e. coords/tangents/points, as long as it matches the expected residual calc arguments,
- `getSample` for `<:AbstractPrior` must return manifold points.

Please update any code you find that does not match.

dehann, 2021-08-03:
Discussion with @Affie lead to...

> Challenge for manifolds on Lie (easier) vs. Riemannian (what ref point, just the mean?).

And suggested PoC on relative factor for S(2) -- how where do we select a reference point and does the 'sampling' happen in the tangent space relative to that point?  From there we can generalize beuyond binary to n-ary

dehann, 2021-08-17:
While talking with @Affie , we think the best way forward is, sampling is based around dispatch on the functions `sample`, and returned elements are **manifold-tangent-vectors** (not points or coordinates):
- `StatsBase.sample(z::Distribution) = rand(z)`
- `StatsBase.sample(mkd::ManifoldKernelDensity, p=identity_element(mkd.manifold)) = log(mkd.manifold, p, __apointsample__)`

Users must then also return a `::Tuple` of the tangent and other meta when overloading `CalcFactor{<:___}`:
- `StatsBase.sample(cf::CalcFactor{<:Pose2Pose2}) = (sample(cf.factor))`
- `StatsBase.sample(cf::CalcFactor{<:ManifoldPrior}) = (sample(cf.factor))`

Affie, 2021-08-18:
Thinking a bit more we can maybe have `getSample(cf::CalcFactor) = (sample(cf.factor.Z),)` in IIF. It can default to `Z` as with `getMesurementParametric`
That way we/users do not have to extend `getSample` if the factor follows the standard design.

Affie, 2021-08-19:
@dehann, also see https://github.com/JuliaManifolds/Manopt.jl/blob/e0ec985b5baf177b5f7d0570899cb66d960f1199/src/helpers/random.jl
It uses `random_point` and `random_tangent`

dehann, 2021-08-21:
I like the separate name `sampleTangent`.  If we standardize to a reduce name later that is fine (maybe in the future it just becomes `sample`).  I doubt there is a short or medium term path for `rand` to become manifold points, so I think `rand` is probably the wrong way to go for us.