JuliaRobotics/RoME.jl #563: Deserialization issue with Point2Point2Range

GearsAD, 2022-03-26:
Trying to deserialize this factor and I'm seeing the issue below.

## MRE

```julia
using DistributedFactorGraphs, IncrementalInference, RoME
using JSON2
point2point2rangeString = "{\"label\":\"x0l3f1\",\"_version\":\"0.18.1\",\"_variableOrderSymbols\":[\"x0\",\"l3\"],\"data\":{\"eliminated\":false,\"potentialused\":false,\"edgeIDs\":[],\"fnc\":{\"Z\":{\"_type\":\"IncrementalInference.PackedNormal\",\"mu\":89.44271909999159,\"sigma\":3.0}},\"multihypo\":[],\"certainhypo\":[1,2],\"nullhypo\":0.0,\"solveInProgress\":0,\"inflation\":5.0},\"tags\":[\"FACTOR\"],\"timestamp\":\"2022-03-26T01:24:44.373-05:00\",\"nstime\":\"0\",\"fnctype\":\"Point2Point2Range\",\"solvable\":1}"
jback = JSON2.read(point2point2rangeString, Dict{String, Any})
f = unpackFactor(fg, jback)
```

## Stack trace
```
Error while unpacking 'x0l3f1' as 'Point2Point2Range', please check the unpacking/packing converters for this factor - 
MethodError: no method matching PackedPoint2Point2Range(; Z=(_type = "IncrementalInference.PackedNormal", mu = 89.44271909999159, sigma = 3))
Closest candidates are:
  PackedPoint2Point2Range(!Matched::PackedSamplableBelief) at /home/samc/.julia/packages/RoME/kRi83/src/factors/Range2D.jl:26 got unsupported keyword argument "Z"
  PackedPoint2Point2Range(!Matched::Any) at /home/samc/.julia/packages/RoME/kRi83/src/factors/Range2D.jl:26 got unsupported keyword argument "Z"
Stacktrace:
  [1] convert(#unused#::Type{PackedPoint2Point2Range}, nt::NamedTuple{(:Z,), Tuple{NamedTuple{(:_type, :mu, :sigma), Tuple{String, Float64, Int64}}}})
    @ DistributedFactorGraphs ~/.julia/dev/DistributedFactorGraphs/src/services/Serialization.jl:353
  [2] convert(#unused#::Type{PackedFunctionNodeData{PackedPoint2Point2Range}}, nt::NamedTuple{(:eliminated, :potentialused, :edgeIDs, :fnc, :multihypo, :certainhypo, :nullhypo, :solveInProgress, :inflation), Tuple{Bool, Bool, Vector{Any}, NamedTuple{(:Z,), Tuple{NamedTuple{(:_type, :mu, :sigma), Tuple{String, Float64, Int64}}}}, Vector{Any}, Vector{Any}, Int64, Int64, Int64}})
    @ DistributedFactorGraphs ~/.julia/dev/DistributedFactorGraphs/src/services/Serialization.jl:357
  [3] unpackFactor(dfg::GraphsDFG{SolverParams, DFGVariable, DFGFactor}, packedProps::Dict{String, Any})
    @ DistributedFactorGraphs ~/.julia/dev/DistributedFactorGraphs/src/services/Serialization.jl:422
```

dehann, 2022-03-26:
Think I know whats wrong here.  Will fix quickly

dehann, 2022-03-26:
`RoME.PackedPoint2Point2Range` is missing the `Base.@kwdef` for auto keyword constructors.  I'm adding it quick

GearsAD, 2022-03-26:
Ah yeah that makes sense, no rush the PR is prepped for the SDK, can do it anytime over the weekend.

dehann, 2022-03-26:
Okay, tested locally and works.  Also added it to the tests with
- #564 

We can tag RoME v0.18.2 after merge!  Apologies for the snag.

closed 2022-03-26 by dehann

GearsAD, 2022-03-26:
Great, thanks!

commit 4b17dc82 (2022-03-26): PackedPoint2Point2Range missing kw constructors,

fixes #563