JuliaRobotics/RoME.jl #641: The use of InertialPose3

saparkison, 2022-12-20:
I'm having trouble figuring out how to use the IntertialPose3 type. Sorry if I am missing an obvious example. I see [/test/testInertialPose3.jl](https://github.com/JuliaRobotics/RoME.jl/blob/master/test/testInertialPose3.jll) but following that approach doesn't seem to work anymore. I get an error with `addVariable!()`
```
ERROR: LoadError: MethodError: no method matching addVariable!(::GraphsDFG{SolverParams, DFGVariable, DFGFactor}, ::Symbol, ::Type{InertialPose3}; N=100)
```
which I think comes down to `InertialPose3 <: InferenceVariable` evaluating as `false`

Am I missing something? Or is there a better example of how to use InertialPose3?

Thanks!

dehann, 2023-01-02:
Hi @saparkison , apologies just noticed your issue now!  The `InertialPose3` factor was withheld on licensing delays, but a full INS integration factor model replacement is in the works using `IncrementalInference.DERelative` factors.  One of big blockers we just cleared late in 2022 was converting the full code base to Manifolds.jl.  The plan is to build out IMU related factors on Manifolds proper.  These will be a bit slower to compute than the preintegration versions, but should have good accuracy and much greater versatility.

Could you perhaps say a bit more on your application to see if there is a stand-in solution we can help find that can work in the short term?

----

The INS-type factor described above is on our roadmap, maybe Q2, Q3 of this year.

Aside: part of the longer build out of the code base here is to be fully consolidated between both parametric and nonparametric solutions (for batch and Bayes tree w/ recycling solutions); while also being fully integrated with Manifolds, ManOpt / Optim, DiffEq, and other packages.

TAC
- [ ] FIXME restore `test/testInertialPose3.jl`

saparkison, 2023-01-04:
Thanks for the response and the explanation. I'll be interested in seeing the `IncrementalInference.DERelative`  implementation and how it compares to preintegration approaches. 

I'm trying to perform state estimation using IMU + GPS data (and maybe eventually wheel encoder data) for a ground vehicle. After looking around more, maybe [/src/services/OdometryUtils.jl](https://github.com/JuliaRobotics/RoME.jl/blob/master/src/services/OdometryUtils.jl) is the place to start looking?

dehann, 2023-09-16:
HI @Affie , FYI see `test/testInertialPose3.jl`

dehann, 2023-11-14:
see related comment:
- https://github.com/JuliaRobotics/RoME.jl/issues/334#issuecomment-1811377786