JuliaRobotics/RoME.jl #692: FileDFG causing loadDFG test error

dehann, 2023-08-08:
Hi @Affie , what is the right way to handle this case between DFGVariable and Variable?

```
Test g2o SE3 export: Error During Test at /home/dehann/.julia/dev/RoME/test/testG2oExportSE3.jl:10
  Got exception outside of a @test
  MethodError: no method matching calcPPE(::Variable; solveKey::Symbol, ppeType::DataType)
  
  Closest candidates are:
    calcPPE(::DFGVariable) got unsupported keyword arguments "solveKey", "ppeType"
     @ IncrementalInference ~/.julia/dev/IncrementalInference/src/services/FGOSUtils.jl:234
    calcPPE(::DFGVariable, ::InferenceVariable; ppeType, solveKey, ppeKey)
     @ IncrementalInference ~/.julia/dev/IncrementalInference/src/services/FGOSUtils.jl:234
    calcPPE(::AbstractDFG, ::Symbol; solveKey, ppeType)
     @ IncrementalInference ~/.julia/dev/IncrementalInference/src/services/FGOSUtils.jl:291

```

Affie, 2023-08-08:
Only DFGVariable supports solver features such as calcPPE. 
Are you having issues with the latest DFG tag? I checked and it loads a `GraphsDFG{NoSolverParams, DFGVariable, DFGFactor}`
On the branch I'm using, I temporarily changed the defaults to load older FileDFGs and save them with all the metadata, so don't use that.

dehann, 2023-08-13:
going to try the fix: `loadDFG!(initfg(), filepath)`

closed 2023-08-14 by dehann