JuliaRobotics/RoME.jl #711: tag

Affie, 2023-10-26:
@JuliaRegistrator register branch=release/v0.24

JuliaRegistrator, 2023-10-26:
Registration pull request created: [JuliaRegistries/General/94160](https://github.com/JuliaRegistries/General/pull/94160)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.24.0 -m "<description of version>" d2c7bdc8cc01e834dea55ef5fa4d6f547c3b5cf2
git push origin v0.24.0
```

Affie, 2023-10-26:
@JuliaRegistrator register branch=release/v0.24

JuliaRegistrator, 2023-10-26:
Registration pull request updated: [JuliaRegistries/General/94160](https://github.com/JuliaRegistries/General/pull/94160)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.24.0 -m "<description of version>" ba0e02908669e5740e0c03fbb764c29eb688fd2d
git push origin v0.24.0
```

closed 2023-10-26 by Affie