JuliaRobotics/RoME.jl #773: tag v0.24.7

dehann, 2025-05-06:
@JuliaRegistrator register branch=release/v0.24

JuliaRegistrator, 2025-05-06:
Registration pull request created: [JuliaRegistries/General/130406](https://github.com/JuliaRegistries/General/pull/130406)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.24.7 -m "<description of version>" d1164a52eb89f88087315e4c7075e8fe071f9ba0
git push origin v0.24.7
```

dehann, 2025-05-06:
@JuliaRegistrator register branch=release/v0.24

Release notes:

- Suppress precompile workload
- Compat for DFG v0.26

JuliaRegistrator, 2025-05-06:
Registration pull request updated: [JuliaRegistries/General/130406](https://github.com/JuliaRegistries/General/pull/130406)

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.24.7 -m "<description of version>" 2d5d9487b1853c90f735ade0cf43ee82fc59714c
git push origin v0.24.7
```

closed 2025-05-06 by dehann